ethz-asl/rovio #8: Using GLSL 3.00 ES

devbharat, 2015-10-20:
Can the shader files be modified to use GLSL version 3.00 ES easily ?
Or can the node be compiled to not require the graphic output used by the shaders ?

Thanks.

bloesch, 2015-10-20:
GLSL 3.3 is required for visualizing the 3D scene. However, you can compile the node without the DMAKE_SCENE flag. Only the image visualization will be supported in that case. Should be enough for most applications...

devbharat, 2015-10-20:
Actually, I made very minor modifications to the shader files to work with GSLS 3.0 ES, it seems to run properly. I can make a PR if you wish.

I am not sure if GLSL 3.3 run on ubuntu with intel graphics.

bloesch, 2015-10-20:
That would be nice. thanks

devbharat, 2015-10-22:
https://github.com/ethz-asl/rovio/pull/12

closed 2015-11-13 by bloesch