ethz-asl/rovio #11: Response to gazebo image stream

devbharat, 2015-10-21:
I am trying to run the ROVIO pipeline on the Imu + Vision stream coming from the gazebo iris model's vi-sensor from rotorSimulator(to eventually use the pose update from ROVIO in the feedback loop with pixhawk SITL).

The estimator seems to diverge for simple hovering datasets in a few seconds. I have updated the .yaml files and the rovio.info file to use the calibration file of the gazebo's camera, and to set the camera-IMU transformation (it's identity). The image scenery is fairly textured (though planar), and the detector finds a tracks features for a while, before the depth uncertainty blows up. 

I have rovio running decently on other "real" vi-sensor datasets, and the one difference I see with the image stream coming out of gazebo is that the framerate varies a lot(see https://github.com/ethz-asl/rotors_simulator/issues/277). Could this cause the estimator to blow up ?

devbharat, 2015-10-22:
The time stamps for both IMU and camera are the gazebosim time, but they run in independent threads.
I'l send you the dataset, it'll be would be great if you could have a look at it.

bloesch, 2015-11-16:
What is the status? Did you manage to make it run in Gazebo? Did you talk with @raghavkhanna? After adapting the extrinsics/intrinsics he got it working pretty quickly in Gazebo...

devbharat, 2015-11-16:
Yes, It works almost perfectly. Here are some plots comparing against ground truth:

Rotation Rates:
[image]
[image]
[image]

Linear Velocities:
[image]
[image]
[image]

Positions:
[image]
[image]
[image]

Except for position in Z, which isn't that good. I saw a little better position tracking in other gazebo bag files, but it was never perfect.

Here is a link to a screen capture video showing the complete setup:
https://vimeo.com/144726358

bloesch, 2015-11-16:
Thanks you very much for the feedback! Especially when looking at the velocity estimates it seems like something could be wrong with the corodinate frames during evaluation. I am not sure which of the rovio outputs you are using for your plots, but some will be with respect to the IMU CF and others will be w.r.t the camera CF. If something is mixed up there it could lead to such behavior.

A student of mine is currently writing a python alignment tool (for rosbags), if you are interested I can give you the link as soon as it is ready.

devbharat, 2015-11-16:
I used rovio/odometry outputs for the plots. Can you tell me which ones are they respect to ? Ideally, we we should be using the ones with respect to IMU coordinate frame for controls, isn't it ? Yes, I'd be interested in the python script! In return, if someone wants to have the px4-sitl-rovio setup to quickly run rovio realtime on a flying quad, I could help set it up.

bloesch, 2015-11-16:
rovio/pose, rovio/output and rovio/odometry express the motion of the camera w.r.t. the inertial frame. rovio/transform is the IMU w.r.t. the inertial frame.

I see that the output is a bit of a mess, i'll try to clean it up, also i will provide the covariance of the output.

bloesch, 2015-11-16:
And yes, the output of the IMU is probably the best to use for control (it is also the one which is actually estimated in ROVIO).

One more thing: did you check out the newest version? There was one version which was less stable/accurate. The current one is definetely better.

devbharat, 2015-11-17:
@bloesch Yes, ploting the ground truth vs rovio/transform removes most of the constant offsets in position and angles! For the linear and angular velocities, the ones published in /rovio/odometry, are they again with respect to Camera CF ? Where should I access the rates wrt the IMU frame ?

Pitch Angle (Red is Ground Truth, Blue is Camera Frame, Green is IMU frame)
[image]

bloesch, 2015-11-17:
/rovio/odometry is currently containing the estimate of the camera CF, I will change this to the IMU CF in the next pull request (today sometime)... then you should be able to access the rates of the IMU

bloesch, 2015-11-18:
Just merged the PR, now the rovio node output should be better. Except for the tf everything is w.r.t. the IMU CF now. Also, if you want to have a look at the post-processing tool please have a look at the following repo:
https://github.com/ethz-asl/trajectory_toolkit
It is not yet fully finished but you can already use it for doing a lot of plots, there is an example.py as well as a rovio specific python script. It should be more or less self explanatory.

devbharat, 2015-11-18:
Thanks! I'll try out the post-processing tool tonight. I will also have rovio in the loop for controls in the simulation(initially only use rovio's position output, then use the full pose).

One last thing, could you tell me if the external pose input (gps/tag-based etc) to rovio's estimator, is that already in master ? Thanks!

bloesch, 2015-11-19:
The exteral pose input support is already in the master branch. Unfortunately its not really well documented. It is turned on a soon as the listeners get messages. Thereis an extra calibration which can be turned on to automatically align the different body frames and inertial frames. There are three modes which you can set by setting the nPose cmake variable. if set to 0 no extra calibratiob is performed. if set to 1 the inertial frames CFs are automatically calibrated. if set to 2 the inertial and body-fixed CFs are calibrated.

I will be on holidays for the next three weeks with limited internet connection. If you have questions regarding the external measurements please contact @raghavkhanna

devbharat, 2015-11-26:
I'll close this issue here, the original problem with gazebo stream(caused by inaccurate Camera CF - IMU CF initialization) is resolved.

closed 2015-11-26 by devbharat

zz1120122573, 2017-12-26:
Hi，I‘m running rovio, but I don't know how to output the pose, could you tell me? Thanks a lot.

anastaga, 2022-07-20:
@devbharat Hello, as i've understood, you used IMU and Camera and managed to synchronized them. Can you share your methodology or link your repo (if there is one)?