ethz-asl/rovio #17: Are any datasets available?

andre-nguyen, 2015-11-07:
Hi,

I was wondering if there were any bag files available, or if any would be made available eventually so that I can try this out and get a feel of how it works.

Thanks

bloesch, 2015-11-08:
We can do this. @burrimi should we provide a link to the euroc datasets?

burrimi, 2015-11-09:
yes sure. The dataset paper finally got accepted and all the datasets are on our dataset page.
http://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets

closed 2015-11-09 by andre-nguyen

andre-nguyen, 2015-11-13:
For anyone stumbling on this issue later, everything works great! I also had a little laugh when I saw this guy pop up from behind a table.

[image]