ethz-asl/rovio #23: Position in inertial-z (altitude) seems to drift

devbharat, 2015-11-19:
When executing vertical motion with the camera facing down, rovio's z estimate seems to drift with respect to gazebo's ground truth with time. It happens more severely when the vertical motion is executed slowly (the slower the worse) and better when accelerations are higher.

I can observe on the Rviz output, that when rising slowly, rather than moving the IMU-Camera position estimate in the upword direction, it adjusts the tracked feature patch's location (then all start condensing closer). This doesn't happen when the vertical accelerations are higher.

Plots below:
Slow vertical motion:
[image]
[image]
[image]

Fast vertical motion:
[image]
[image]
[image]

bloesch, 2015-11-19:
Hmmm, interesting, can you observe that effect on real data as well? you could try increasing the noise on the predicted position (in the info file).

devbharat, 2015-11-19:
I do not have real datasets with such vertical motions (we will have some in early december). I'll try to play around with the noise parameters you mention (and others) and try to see why exactly this happens. To me it appears that when the vertical accelerations are low, for example in almost quasi-static climb, the estimator seems to trust the IMU too much (whose measurements are lost in the noise) and compensates for the change in camera image by adjusting the positions of the tracked features. Of course then it breaks down when I climb back down into the scene, almost right through the estimated feature positions.

devbharat, 2015-11-26:
This does not happen anymore as of commit b1fd00886b731 .

closed 2015-11-26 by devbharat