ethz-asl/rovio #24: Pose Update from Tag Detector

devbharat, 2015-11-26:
I have been trying to include the raw pose obtained from tag detector into rovio. Both Tag based position and attitude are being used, and pose feedback to features in rovio is enabled. The problem is that even though the raw tag based position follows the ground truth pretty well (see plots below), the rovio output does not converge to it, even after multiple measurements.

[image]

My rovio.info file looks like:
nPose was set to 1.

    PoseUpdate
    {
        UpdateNoise
        {
            pos_0 0.5;							X-Covariance of linear pose measurement update [m^2]
            pos_1 0.5;							Y-Covariance of linear pose measurement update [m^2]
            pos_2 1e-7;							Z-Covariance of linear pose measurement update [m^2]
            att_0 0.5;							X-Covariance of rotational pose measurement update [rad^2]
            att_1 0.5;							Y-Covariance of rotational pose measurement update [rad^2]
            att_2 0.5;							Z-Covariance of rotational pose measurement update [rad^2]
        }
    	init_cov_IrIW 10;							Covariance of initial pose between inertial frames, linear part [m^2]
    	init_cov_qWI 10;								Covariance of initial pose between inertial frames, rotational part [rad^2]
    	init_cov_MrMV 10;							Covariance of initial pose between inertial frames, linear part [m^2]
    	init_cov_qVM 10;								Covariance of initial pose between inertial frames, rotational part [rad^2]
        pre_cov_IrIW 1e-4;							Covariance parameter of pose between inertial frames, linear part [m^2/s]
        pre_cov_qWI 1e-4;							Covariance parameter of pose between inertial frames, rotational part [rad^2/s]
        pre_cov_MrMV 1e-4;							Covariance parameter of pose between inertial frames, linear part [m^2/s]
        pre_cov_qVM 1e-4;							Covariance parameter of pose between inertial frames, rotational part [rad^2/s]
        MahalanobisTh0 12.6511204;					Mahalonobis distance treshold of pose measurement
    	doVisualization true;						Should the measured poses be vizualized
    	enablePosition true;						Should the linear part be used during the update
    	enableAttitude true;						Should the rotation part be used during the update (e.g. set to fals eif feeding GPS-measurement)
    	noFeedbackToRovio false;						By default the pose update is only used for aligning coordinate frame. Set to false if ROVIO should benefit frome the posed measurements.
    	doInertialAlignmentAtStart true;			Should the transformation between I and W be explicitly computed and set with the first pose measurement.
    	timeOffset 0.0;								Time offset added to the pose measurement timestamps
        qVM_x 0;									X-entry of quaterion representing IMU to reference body coordinate frame of pose measurement (JPL)
        qVM_y 0;									Y-entry of quaterion representing IMU to reference body coordinate frame of pose measurement (JPL)
        qVM_z 0;									Z-entry of quaterion representing IMU to reference body coordinate frame of pose measurement (JPL)
        qVM_w 1;									W-entry of quaterion representing IMU to reference body coordinate frame of pose measurement (JPL)
        MrMV_x 0;									X-entry of vector representing IMU to reference body coordinate frame of pose measurement
        MrMV_y 0;									Y-entry of vector representing IMU to reference body coordinate frame of pose measurement
        MrMV_z 0;									Z-entry of vector representing IMU to reference body coordinate frame of pose measurement
        qWI_x 0;									X-entry of quaterion representing World to reference inertial coordinate frame of pose measurement (JPL)
        qWI_y 0;									Y-entry of quaterion representing World to reference inertial coordinate frame of pose measurement (JPL)
        qWI_z 0;									Z-entry of quaterion representing World to reference inertial coordinate frame of pose measurement (JPL)
        qWI_w 1;									W-entry of quaterion representing World to reference inertial coordinate frame of pose measurement (JPL)
        IrIW_x 0;									X-entry of vector representing World to reference inertial coordinate frame of pose measurement
        IrIW_y 0;									Y-entry of vector representing World to reference inertial coordinate frame of pose measurement
        IrIW_z 0;									Z-entry of vector representing World to reference inertial coordinate frame of pose measurement
    }

The same configuration when used with not raw tag based pose, but with ground truth from gazebo, I get excellent results (even with these horribly high covariance values):

Result with sending ground truth position to rovio:
[image]

Essentially, the only difference I see between the position updates from the tag versus that from ground truth is not the position value itself (although the tag based position is slightly noisy), but the **time delay** of the detector actually processing the image to extract the pose(like 300 ms).

I also constantly get the warning:

    Warning: included update measurements before safeTime

when including tag based pose which I don't get when using ground truth. 

Also, the visualization of ground truth on the 'scene' window of ROVIO is not smooth when using tag based pose (it momentarily moves then stays put even as rovio's CF moves around), but very smooth (and agreeing with ROVIO's CF) when using ground truth pose.

mhkabir, 2015-11-30:
@devbharat What tag detector are you using?

mhkabir, 2015-11-30:
Also -  What do you use for those sweet looking plots?

bloesch, 2015-12-05:
the filter in rovio has a maximal duration for waiting for a certain measurement after which it will just continue with the estimation and ignore any measurements which come to late (thus the warning). this duration can be adapted but is for the moment hardcoded. you should be able to find 'waitingTime' as a member of the measurement timeline which itself should be a member of RovioFilter (please have a look at the Filter.hpp in lightweightfiltering).

devbharat, 2015-12-16:
A couple of different approaches were discussed to deal with this scenario of getting delayed pose measurements:

 - Just waiting longer in rovio by increasing the _maxWaitingTime in the RovioFilter as described above. This would cause the estimator output to be delayed and 'may' cause instabilities in the feedback loop.
 - Same like above, running rovio's internal state in a delayed manner, but forward integrating the state through the IMU measurements to the current time. If not done for 'too long' delay, this shouldn't drift much. Also, not very computationally expensive and not too complex implementation wise. Could possibly be implemented in rovio.
- The 'statistically clean' way, you clone the 'appropriate state' where you expect the pose measurement to arrive, and augment it to your filter state. When the pose measurement arrives, apply it to the previous cloned state. Since it's augmented in the filter state, the information will 'flow' to the latest state too.