ethz-asl/rovio #25: The various coordinate frames used in ROVIO

devbharat, 2015-11-26:
It'd be great if we had a description of the various coordinate frames used in ROVIO and the acronyms used to refer to them in the code.

We have the 
IMU CF:
Located at IMU origin and axis aligned along IMU's internal accelerometers.
Referred with letter **M** in the rovio.info file. 

Camera CF:
Located at camera optic-center (or at the top-left corner of the image plane ?) and axis along the image plane x and y axis and z going out of the image plane. 
Referred with letter **C** in the rovio.info file. 

Reference Body CF:
Referred with letter **V** in the rovio.info file. 
 
Reference Inertial CF:
Referred with letter **I** in the rovio.info file. 

World CF:
Referred with letter **W** in the rovio.info file.

I created a rough diagram from what I got in the discussion with @raghavkhanna and @bloesch earlier: 

[image]

Correct me where I am wrong.

mhkabir, 2015-11-30:
@devbharat 
I'm trying to get a setup running with Rovio. This just a standard IMU-synced forward-facing stereo rig, with a 11 degrees downward tilt. Not a VI-sensor or anything. 
Now I'm really having a difficulty getting all the coordinate transforms in my head. Could someone help me out figuring out qCM?

The IMU is mounted flat on the body, and the cameras are forward mounted.

raghavkhanna, 2015-12-02:
@devbharat 
I agree. I would encourage you to add this to the readme and send in a pull request. Here are a few pointers (@bloesch feel free to correct me if I'm wrong)

1. Camera CF origin is located at the principal point.

Everything else looks good to me.

bloesch, 2015-12-05:
Everything seems correct, thanks for putting this together. optic-center is correct. i will check the rest of the code such that everything is consistent eith this notation (i used a different notatiob at the beginning).

@mhkabir: qCM should describe the rotation between camera and imu. while this gets co-estimated online the initial guess should be to wrong ( depending on the parameters and actual motion you might be able to correct 10-90° error). what you could try is to enable the verbose in the info file and test different 90° rotations between the coordinate frame. if one seems do end up in a stable estimation (after having excited/moved the sensor) you can read the estimated calibration from the shell output.

kingfolk, 2015-12-09:
Hi, all. I am trying to setup the CF transformation between IMU and camera too. According to @bloesch recommendation, i try lots of combinations of pitch roll and yaw. But the tf in scene window fall out of the viewport... I also notice there are two imu frame, 'M' and 'I'. Do i also need to set 'I' in rovio.info?, if ROVIO also take imu orientation into account.
 

thanks in advance.

bloesch, 2015-12-10:
Two comments which might be helpfull: the view in the scene can be changed with the mouse or with the keyboard (arrows, page up, page down). Also, the 'I' coordinate frame does not relate to the IMU frame but to an addional inertial coordinate frame which is only used if you add optional pose measurement (e.g. GPS).

I will try to write a clearer introduction to ROVIO and how to set things up. Typically the extrinsic calibration qCB/BrBC are the most critical parts and it is difficult to provide people with the correct values without having a look at the hardware and specification (for instance every IMU manufacturer typically provides information on the orientation and locations of their reference frames). Other parts which can cause problems are bad timestamps (they should all be expressed with respect to a same reference time), intrinsic calibration (which have to be set in the yaml file) or noise parameters (although the default values should often yield okeisch results).

bloesch, 2015-12-14:
I started a wiki trying to clarifying a few points, please let me know if things are unclear or if further information are required. @devbharat adding a diagram would probably be helpful, do you mind if I add the one you did above? I also just saw a little error: I-W is also calibrated if nPose is set to 2.

devbharat, 2015-12-14:
Great! Yes you can use it!

devbharat, 2015-12-16:
Some other options as discussed could be:
  - Auto-detecting the closest 90 degree orientation by running test passes through recorded bag files with all possible 90 degree rotations
  - Reading Camera-Imu SE3 transformation from the yaml file generated by kalibr