ethz-asl/rovio #26: Packing use of visualization_msgs with MAKE_SCENE

devbharat, 2015-12-02:
I am porting rovio embedded linux using  yocto and meta-ros, and it seems wasteful to have rovio sending (and requiring to be built with) visualization_msgs when it's not built with scene, and linux built without support for GUI. 

I can manually edit it out, but I was wondering if it made sense in general to send out visual marker messages only if rovio is built with scene.

bloesch, 2015-12-05:
i am no completely sure if i get it write. the build with scene is meant for the opengl visualization. indepent of that someone can use the visualisation_msgs within rviz or for some other purposes. also the visualization shouldn t be computed and published if their are no listeners.

devbharat, 2015-12-16:
This is actually good. The markers are actually helpful when you need to visualize when you don't build rovio with scene.

closed 2015-12-16 by devbharat