ethz-asl/rovio #27: Running on embedded linux

devbharat, 2015-12-03:
So I have rovio finally running on IMX6 on embedded linux. It has a quad core 1GHz processor, but rovio uses only 1 core, and I get 3-4 frames per second.

I see that currently the CMake file disables OpenMP by default, has someone tried it running parallelly ?

devbharat, 2015-12-03:
With it enabled, I get what seem like memory access related errors:

    [ 1516.679816] 1812 total pagecache pages
    [ 1516.683582] 0 pages in swap cache
    [ 1516.686903] Swap cache stats: add 0, delete 0, find 0/0
    [ 1516.692130] Free swap  = 0kB
    [ 1516.695024] Total swap = 0kB
    [ 1516.720815] 262144 pages of RAM
    [ 1516.723991] 1340 free pages
    [ 1516.726791] 7042 reserved pages
    [ 1516.729935] 2675 slab pages
    [ 1516.732746] 266383 pages shared
    [ 1516.735892] 0 pages swap cached
    [ 1516.739038] [ pid ]   uid  tgid total_vm      rss nr_ptes swapents oom_score_adj name
    [ 1516.746907] [  120]     0   120     2194      315       7        0             0 systemd-journal
    [ 1516.755716] [  149]   996   149     3077       50       6        0             0 systemd-timesyn
    [ 1516.764524] [  158]     0   158     2391       99       5        0         -1000 systemd-udevd
    [ 1516.773158] [  176]     0   176      713       62       4        0             0 systemd-logind
    [ 1516.781867] [  177]   994   177      794       54       4        0             0 avahi-daemon
    [ 1516.790413] [  178]   998   178      713       63       5        0          -900 dbus-daemon
    [ 1516.798874] [  193]   994   193      794       42       4        0             0 avahi-daemon
    [ 1516.807421] [  213]   988   213      692       47       5        0             0 systemd-network
    [ 1516.816227] [  214]     0   214      650       18       5        0             0 klogd
    [ 1516.824177] [  233]     0   233      666       15       5        0             0 syslogd
    [ 1516.832280] [  262]     0   262      909       74       5        0             0 login
    [ 1516.840229] [  271]     0   271      908       73       6        0             0 login
    [ 1516.848172] [  435]   987   435      783       41       5        0             0 systemd-resolve
    [ 1516.856980] [  438]     0   438      866       72       4        0             0 systemd
    [ 1516.865091] [  440]     0   440     1332      262       5        0             0 (sd-pam)
    [ 1516.873291] [  442]     0   442      768       80       5        0             0 sh
    [ 1516.880958] [  443]     0   443    11091     2245      17        0             0 python
    [ 1516.888983] [  456]     0   456    19869     2102      21        0             0 python
    [ 1516.897008] [  469]     0   469    11396      146      14        0             0 rosout
    [ 1516.905037] [  575]     0   575      757       76       6        0             0 sh
    [ 1516.912717] [  587]     0   587      909       74       5        0             0 login
    [ 1516.920646] [  589]     0   589      757       76       6        0             0 sh
    [ 1516.928333] [ 1242]     0  1242      909       74       6        0             0 login
    [ 1516.936281] [ 1247]     0  1247      757       76       4        0             0 sh
    [ 1516.943971] [ 1458]     0  1458    10970     2203      16        0             0 python
    [ 1516.951995] [ 1476]     0  1476   292033   195125     568        0             0 rovio_node
    [ 1516.960380] [ 1526]     0  1526     3895     1960      13        0             0 python
    [ 1516.968417] [ 1529]     0  1529    13925     2096      19        0             0 play
    [ 1516.976279] [ 1546]     0  1546     3895     1961      11        0             0 python
    [ 1516.984315] [ 1549]     0  1549    37460      888      31        0             0 record
    [ 1516.992336] [ 1668]     0  1668      731       32       5        0             0 top
    [ 1517.000116] Out of memory: Kill process 1476 (rovio_node) score 744 or sacrifice child

devbharat, 2015-12-04:
It looks like there is some memory management issue in ROVIO, like memory allocated in heap what is never freed when out of use, as I can see all crashes are actually caused as the memory consumed by the process monotonically increases until it consumes all there is on ROVIO and then crashes.

bloesch, 2015-12-05:
It has been a while that I ve been testing openmp. It used to work but since I mainly got higher cpu load without significant performance increase (maybe 20%), I ended up disabling it by default.

Not sure about the memory issue, I testing things with valgrind, and apart from some ros/vizualization memory issues I couldnt find any within rovio itself.

Maybe have a look at the openmp on a regular computer first.

Also, if you want to get higher frame rate try to reduce the number if features. things should still work properly with 10 features. adapting some other parameters you could probably even reduce it further and still get an acceptable performance.

devbharat, 2015-12-16:
I think the issue is more to do with ROS than rovio. The RAM consumption remains limited (to around 200 mb) as log as I run the bag files at low rate (like 20% real time), but keeps increasing if I play the bag file faster.

Could it be that rovio caches the images and imu data it is yet to process, and this cache keeps growing bigger and bigger (or does it begin to drop frames if cache size reaches some limit ?).

devbharat, 2015-12-16:
Also, In the bag files I record onboard the embedded system, I can see 'delay' in the rovio output compared to the ground truth, which is weird because even if rovio takes longer to compute, and file IO is probably slower too, the timestamp of the computed output should be the same as that of the raw_image associated with the output, so it shouldn't "lag" behind the ground truth in the plots, which it does, more and more for bag files played faster.

[image]

devbharat, 2015-12-16:
Regarding the 'Delay', this should not happen. Could it be:
  - Check whether the time-stamps are properly managed in ROVIO
  - Check if it isn't a post-processing timestamp issue.
  - ?

Regarding reducing the system load, multiple approaches to try within rovio:
  - Reduce feature count, 15 is 'ok'
  - Reduce number of levels in the image - pyramid
  - Disable patch-warping 
  - Use 6x6 patches instead of 8x8 patches
  - set alignMaxUniSample to zero to not distribute samples when uncertainty increases 

Regarding increased RAM when bag file is played faster:
  - Seems like a cache issue, is rovio storing all unprocessed data ?
  - Check how callbacks are handled in ros.

bloesch, 2015-12-17:
Thanks a lot for summarizing this. I think you could probably even reducde the patch size to 4x4 for very time critical applications. I will start a wiki-page for keeping a couple of notes on computational costs.

ginowu, 2016-03-23:
@devbharat , I had run one medium level dataset on snapdragon 801 with one core, and can run 14 frames per second with 15 frames. For your test result charts, could you help elaborate on how do you extract the ground truth data in the bag files and compare to the estimated result?

Thanks in advance.

chengshy, 2016-03-29:
@bloesch The queue size for image subscriber is set to 1000. I think when the embedded system has too much computation load, it will keep processing the delayed image. However, if we want to test drone fly with ROVIO, we would rather to drop images to get current pose. Is the RAM problem also because of this setting?

mhkabir, 2016-03-29:
@chengshy As an experiment you could try to drop the queue size to around 10 or so.

chengshy, 2016-03-29:
@mhkabir The delay issue seems be right after I change the image queue size to 1 and imu queue size to 4 ( I just want to test the latest data). But this change may break the performance of the tracker

andre-nguyen, 2016-03-29:
This brings an interesting point. What's better between having less data but higher accuracy settings versus more data but lower accuracy settings.

bloesch, 2016-03-30:
A couple of comments:

@ginowu: there is a python based evaluation tool which is available under [https://github.com/ethz-asl/trajectory_toolkit](https://github.com/ethz-asl/trajectory_toolkit). Unfortunately its an old version. I'll try to update it at some point. Made a new issue.

Concerning the queue: reducing the length of the image queue definetely makes sense. However, you should probably keep the IMU queue a bit longer. This shouldn't affect the computational costs too strongly and could improve robustness.