ethz-asl/rovio #29: Building with OpenCV3

mhkabir, 2015-12-08:
I'm trying to build Rovio with OpenCV 3, which I installed from source. I fixed a couple of the smaller issues which gets me to this stage : 
```
[ 14%] [ 28%] Building CXX object CMakeFiles/rovio.dir/src/rovio_node.cpp.o
Building CXX object CMakeFiles/rovio.dir/src/FeatureCoordinates.cpp.o
In file included from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/MultilevelPatch.hpp:34:0,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/FeatureManager.hpp:35,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/FilterStates.hpp:38,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/RovioFilter.hpp:34,
                 from /home/artemis/artemis_ws/src/vision/rovio/src/rovio_node.cpp:31:
/home/artemis/artemis_ws/src/vision/rovio/include/rovio/ImagePyramid.hpp: In instantiation of void rovio::ImagePyramid<n_levels>::detectFastCorners(std::vector<rovio::FeatureCoordinates>&, int, int) const [with int n_levels = 4]:
/home/artemis/artemis_ws/src/vision/rovio/include/rovio/ImgUpdate.hpp:906:11:   required from void rovio::ImgUpdate<FILTERSTATE>::commonPostProcess(rovio::ImgUpdate<FILTERSTATE>::mtFilterState&, const mtMeas&) [with FILTERSTATE = rovio::FilterState<25u, 4, 8, 2, 0>; rovio::ImgUpdate<FILTERSTATE>::mtFilterState = rovio::FilterState<25u, 4, 8, 2, 0>; rovio::ImgUpdate<FILTERSTATE>::mtMeas = rovio::ImgUpdateMeas<rovio::State<25u, 4, 8, 2, 0> >]
/home/artemis/artemis_ws/src/vision/rovio/include/rovio/ImgUpdate.hpp:706:41:   required from void rovio::ImgUpdate<FILTERSTATE>::postProcess(rovio::ImgUpdate<FILTERSTATE>::mtFilterState&, const mtMeas&, const mtOutlierDetection&, bool&) [with FILTERSTATE = rovio::FilterState<25u, 4, 8, 2, 0>; rovio::ImgUpdate<FILTERSTATE>::mtFilterState = rovio::FilterState<25u, 4, 8, 2, 0>; rovio::ImgUpdate<FILTERSTATE>::mtMeas = rovio::ImgUpdateMeas<rovio::State<25u, 4, 8, 2, 0> >; rovio::ImgUpdate<FILTERSTATE>::mtOutlierDetection = rovio::ImgOutlierDetection<rovio::State<25u, 4, 8, 2, 0> >]
/home/artemis/artemis_ws/src/vision/rovio/src/rovio_node.cpp:118:1:   required from here
/home/artemis/artemis_ws/src/vision/rovio/include/rovio/ImagePyramid.hpp:137:75: error: no matching function for call to cv::FastFeatureDetector::FastFeatureDetector(int&, bool)
     cv::FastFeatureDetector feature_detector_fast(detectionThreshold, true);
                                                                           ^
/home/artemis/artemis_ws/src/vision/rovio/include/rovio/ImagePyramid.hpp:137:75: note: candidates are:
In file included from /usr/local/include/opencv2/features2d/features2d.hpp:48:0,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/FeatureCoordinates.hpp:34,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/RobocentricFeatureElement.hpp:33,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/CoordinateTransform/FeatureOutput.hpp:34,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/FilterStates.hpp:36,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/RovioFilter.hpp:34,
                 from /home/artemis/artemis_ws/src/vision/rovio/src/rovio_node.cpp:31:
/usr/local/include/opencv2/features2d.hpp:411:20: note: cv::FastFeatureDetector::FastFeatureDetector()
 class CV_EXPORTS_W FastFeatureDetector : public Feature2D
                    ^
/usr/local/include/opencv2/features2d.hpp:411:20: note:   candidate expects 0 arguments, 2 provided
/usr/local/include/opencv2/features2d.hpp:411:20: note: cv::FastFeatureDetector::FastFeatureDetector(const cv::FastFeatureDetector&)
/usr/local/include/opencv2/features2d.hpp:411:20: note:   candidate expects 1 argument, 2 provided
/usr/local/include/opencv2/features2d.hpp:411:20: note: cv::FastFeatureDetector::FastFeatureDetector(cv::FastFeatureDetector&&)
/usr/local/include/opencv2/features2d.hpp:411:20: note:   candidate expects 1 argument, 2 provided
In file included from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/MultilevelPatch.hpp:34:0,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/FeatureManager.hpp:35,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/FilterStates.hpp:38,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/RovioFilter.hpp:34,
                 from /home/artemis/artemis_ws/src/vision/rovio/src/rovio_node.cpp:31:
/home/artemis/artemis_ws/src/vision/rovio/include/rovio/ImagePyramid.hpp:137:29: error: cannot declare variable feature_detector_fast to be of abstract type cv::FastFeatureDetector
     cv::FastFeatureDetector feature_detector_fast(detectionThreshold, true);
                             ^
In file included from /usr/local/include/opencv2/features2d/features2d.hpp:48:0,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/FeatureCoordinates.hpp:34,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/RobocentricFeatureElement.hpp:33,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/CoordinateTransform/FeatureOutput.hpp:34,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/FilterStates.hpp:36,
                 from /home/artemis/artemis_ws/src/vision/rovio/include/rovio/RovioFilter.hpp:34,
                 from /home/artemis/artemis_ws/src/vision/rovio/src/rovio_node.cpp:31:
/usr/local/include/opencv2/features2d.hpp:411:20: note:   because the following virtual functions are pure within cv::FastFeatureDetector:
 class CV_EXPORTS_W FastFeatureDetector : public Feature2D
                    ^
/usr/local/include/opencv2/features2d.hpp:424:26: note: 	virtual void cv::FastFeatureDetector::setThreshold(int)
     CV_WRAP virtual void setThreshold(int threshold) = 0;
                          ^
/usr/local/include/opencv2/features2d.hpp:425:25: note: 	virtual int cv::FastFeatureDetector::getThreshold() const
     CV_WRAP virtual int getThreshold() const = 0;
                         ^
/usr/local/include/opencv2/features2d.hpp:427:26: note: 	virtual void cv::FastFeatureDetector::setNonmaxSuppression(bool)
     CV_WRAP virtual void setNonmaxSuppression(bool f) = 0;
                          ^
/usr/local/include/opencv2/features2d.hpp:428:26: note: 	virtual bool cv::FastFeatureDetector::getNonmaxSuppression() const
     CV_WRAP virtual bool getNonmaxSuppression() const = 0;
                          ^
/usr/local/include/opencv2/features2d.hpp:430:26: note: 	virtual void cv::FastFeatureDetector::setType(int)
     CV_WRAP virtual void setType(int type) = 0;
                          ^
/usr/local/include/opencv2/features2d.hpp:431:25: note: 	virtual int cv::FastFeatureDetector::getType() const
     CV_WRAP virtual int getType() const = 0;

```
Any help with getting it running would be great! Thanks!

bloesch, 2015-12-14:
Apparently the cv::FastFeatureDetector constructor does not take any arguments anymore. Please have a look at the documentation (http://docs.opencv.org/3.0-rc1/df/d74/classcv_1_1FastFeatureDetector.html) for adapting the constructor.

bloesch, 2016-01-14:
@mhkabir any news on this issue? did you manage to make it work with opencv 3?

TSC21, 2016-01-14:
@bloesch check #32. Got it working now!

bloesch, 2016-01-18:
Thanks for looking into this.

@TSC21: I will try it locally on my machine at some point. I think that the master should be kept with opencv 2.4 since most users will have it installed trough ROS. I will make a branch for the opencv 3 version.

mhkabir, 2016-01-18:
We don't actually need to make a separate branch. I already fixed this in a portable way. Standby for PR.

TSC21, 2016-01-18:
I will add the version verification ao it can pick the code based on the version being used by the user. So as @mhkabir says, no need to add a separate branch.
@mhkabir I would be more useful if you could show a video test of this against your system, since I don't have a camera with a trigger input. Thanks

kartikmohta, 2016-05-23:
This is fixed in b0a42ba8e1f5428a15211fc771c9a6e1d08ff6db (Merge of #52).

closed 2016-05-24 by mhkabir