ethz-asl/rovio #30: Increase robustness to moving objects

burrimi, 2015-12-14:
When we have something moving in front of the camera, the accurracy is really bad.

bloesch, 2015-12-17:
In the current default configuration the outlier detection is very limited. This has something to do with the pre-alignment of the patches where the predicted innovation matrix does not properly represent the noise of the innovation itself. I think the could be fixed somehow, but probably adding some dedicated outlier rejection should also be considered at some point.

bloesch, 2016-03-10:
This was improved in the last pull request. Different parameters to make the rejection stricter or not. Should be adapted to the specific scenario.

closed 2016-09-19 by bloesch

suyash023, 2021-05-04:
Hello! I know that this is a closed issue but can someone please tell me what are the paramters to look out for when tuning ROVIO to be robust to moving objects in the environment? I am aware that mahalanobisThreshold in the info is one such parameter. Can someone please point me to other parameters as well?
Thanks!