ethz-asl/rovio #33: file INSTALL cannot copy file 'lightweight_filtering/cmake/FindLWF.cmake'

TSC21, 2016-01-14:
I noticed that after running `catkin build --cmake-args -DCMAKE_BUILD_TYPE=Release -DMAKE_SCENE=ON` (or just raw `catkin build`), I get an install error bellow due too restricted permissions. `sudo` prefix solves it but that is not recommended. Is this supposed to happen?
```bash
nuno@nuno-P17SM:~/AIMAV_Project$ catkin build --cmake-args -DCMAKE_BUILD_TYPE=Release -DMAKE_SCENE=ON
--------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/nuno/AIMAV_Project/devel:/home/nuno/AIMAV_Project/devel_isolated/rovio:/home/nuno/AIMAV_Project/devel_isolated/kindr:/home/nuno/PINCAir/src/visim_drone/ws/devel:/opt/ros/jade
Workspace:                   /home/nuno/AIMAV_Project
Source Space:       [exists] /home/nuno/AIMAV_Project/src
Build Space:        [exists] /home/nuno/AIMAV_Project/build
Devel Space:        [exists] /home/nuno/AIMAV_Project/devel
Install Space:     [missing] /home/nuno/AIMAV_Project/install
DESTDIR:                     None
--------------------------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            True
Isolate Installs:            False
--------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DMAKE_SCENE=ON
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------------------------- 
Found '2' packages in 0.0 seconds. 
Starting ==> kindr                                                              
Finished <== kindr [ 0.8 seconds ]                                              
Starting ==> rovio                                                              
                                                                                
[rovio] ==> '/home/nuno/AIMAV_Project/build/rovio/build_env.sh /usr/bin/make install' in '/home/nuno/AIMAV_Project/build/rovio'
[ 62%] Built target rovio
[ 75%] Built target feature_tracker_node
[ 87%] Built target rovio_node
[100%] Built target rovio_rosbag_loader
Install the project...
-- Install configuration: "Release"
-- Installing: /home/nuno/AIMAV_Project/install/_setup_util.py
-- Installing: /home/nuno/AIMAV_Project/install/env.sh
-- Installing: /home/nuno/AIMAV_Project/install/setup.bash
-- Installing: /home/nuno/AIMAV_Project/install/setup.sh
-- Installing: /home/nuno/AIMAV_Project/install/setup.zsh
-- Installing: /home/nuno/AIMAV_Project/install/.rosinstall
-- Installing: /home/nuno/AIMAV_Project/install/lib/pkgconfig/rovio.pc
-- Installing: /home/nuno/AIMAV_Project/install/share/rovio/cmake/rovioConfig.cmake
-- Installing: /home/nuno/AIMAV_Project/install/share/rovio/cmake/rovioConfig-version.cmake
-- Installing: /home/nuno/AIMAV_Project/install/share/rovio/package.xml
-- Up-to-date: /home/nuno/AIMAV_Project/install/_setup_util.py
-- Up-to-date: /home/nuno/AIMAV_Project/install/env.sh
-- Up-to-date: /home/nuno/AIMAV_Project/install/setup.bash
-- Up-to-date: /home/nuno/AIMAV_Project/install/setup.sh
-- Up-to-date: /home/nuno/AIMAV_Project/install/setup.zsh
-- Up-to-date: /home/nuno/AIMAV_Project/install/.rosinstall
-- Installing: /home/nuno/AIMAV_Project/install/lib/pkgconfig/lightweight_filtering.pc
-- Installing: /home/nuno/AIMAV_Project/install/share/lightweight_filtering/cmake/lightweight_filteringConfig.cmake
-- Installing: /home/nuno/AIMAV_Project/install/share/lightweight_filtering/cmake/lightweight_filteringConfig-version.cmake
-- Installing: /home/nuno/AIMAV_Project/install/share/lightweight_filtering/package.xml
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/ModelBase.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/TestClasses.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/Update.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/PropertyHandler.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/SigmaPoints.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/Prediction.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/common.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/FilterBase.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/CoordinateTransform.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/State.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/OutlierDetection.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/FilterState.hpp
-- Installing: /home/nuno/AIMAV_Project/install/include/LWF/include/lightweight_filtering/GIFPrediction.hpp
-- Installing: /usr/share/cmake-3.2/Modules/FindLWF.cmake
CMake Error at lightweight_filtering/cmake_install.cmake:157 (file):
  file INSTALL cannot copy file
  "/home/nuno/AIMAV_Project/src/rovio/lightweight_filtering/cmake/FindLWF.cmake"
  to "/usr/share/cmake-3.2/Modules/FindLWF.cmake".
Call Stack (most recent call first):
  cmake_install.cmake:134 (include)

make: *** [install] Error 1
[rovio] <== '/home/nuno/AIMAV_Project/build/rovio/build_env.sh /usr/bin/make install' failed with return code '2'
                                                                                 
Failed   <== rovio [ 1 minute and 24.4 seconds ]                                
[build] There were '1' errors:                                                  

Failed to build package 'rovio' because the following command:

# Command to reproduce:
cd /home/nuno/AIMAV_Project/build/rovio && /home/nuno/AIMAV_Project/build/rovio/build_env.sh /usr/bin/make install; cd -

# Path to log:
cat /home/nuno/AIMAV_Project/build/build_logs/rovio.log

Exited with return code: 2 
```

bloesch, 2016-01-14:
I am not completely sure, but for some reason your install flag is set to true "Install Packages:            True". The issue is that in this case LWF will be installed on your system which requires sudo.

TSC21, 2016-01-14:
Yeah I removed and created my ws again and it solved the issue. Don't know why I had that flag up, even after `catkin clean --all`. Closing...

closed 2016-01-14 by TSC21