ethz-asl/rovio #35: Publishing visualiser output over ROS

mhkabir, 2016-01-23:
I'm running Rovio on a small MAV, with a high-bandwidth link to the GCS, and it is preferable to have the visualiser window display published over a ROS topic. I can already turn off the visualiser window using the requisite param, but being able to see Rovio's output in e.g Rviz (over the network) on the ground would be very useful.

Could someone give me some pointers on this? I'm willing to implement it and send in a PR.

andre-nguyen, 2016-01-23:
I'm guessing its as simple as changing these lines 
https://github.com/ethz-asl/rovio/blob/afe562f1d3d91a1a5d6ed23fbb3d1b735cfc1e3c/include/rovio/RovioNode.hpp#L423

mhkabir, 2016-01-24:
Awesome. I don't know how I even missed that :P

bloesch, 2016-01-24:
you'll just have to add a publisher to rovioNode and publish the image in "mpFilter_->safe_.img_[i]"

mhkabir, 2016-01-24:
Yep, done it. Just give me a day to test this and put up a PR.

closed 2016-09-19 by bloesch

svisschers, 2018-03-02:
I hope you can answer this quickly:

I am trying to implement this solution but I cannot find which encoding is used for the visualiser output. Therefore, when I try to create a ROS message, I get an error due to an encoding mismatch.

Can any of you help me solve this problem?