ethz-asl/rovio #36: camera draft away with low level IMU module

ayumizll, 2016-01-25:
We want run ROVIO with our IMU (mpu6050 200Hz) module and camera(logitech c525, 30Hz) instead of provided dateset, and we followed the instruction noted in https://github.com/ethz-asl/rovio/wiki/Configuration, particularly We changed the IMU-Camera configration as figure shown below.
[image]

For test we fixed our device to be stationary on table, we assured it without any movement, but it draft away unexpectedly. We uploaded the result and you may check it with this link https://youtu.be/mDRDnI2Tam4.

Would you like to give me some advices what should I do to fix it out.

Looking forward to your reply! 

Merci beaucoup.

bloesch, 2016-01-25:
Does the pose drift at the start or only after a certain time? Since you are using a monocular setup rovio will not be able to estimate the distance of the feature, which reduces the amount of information which can be estracted and eventually leads to drift. Do you have a dataset with motion?

Also, you should probably adapted the noise parameter of rovio to your IMU specifications. Please adapt vel (velocity random walk), acb (accelerometer bias random walk), gyb (gyroscope bias random walk), att (angular random walk).

Other reasons for the drift could be: bad timings, bad initial guess for the extrinsics, or innacurate intrinsic calibration.

Also: what is the output of your accelerometer, m/s^2? could you print a message and check if the norm of the accelerometer reading is close to 9.81?

mhkabir, 2016-01-25:
@bloesch Does Rovio follow similar IMU specifications as Kalibr? https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model-and-Intrinsics

bloesch, 2016-01-25:
yes. just note that the values in vel/acb/att/gyb are the squares of the standard deviations or sigmas used in kalibr.

ayumizll, 2016-01-25:
Thanks a lot. 
Drift happens from the begining of time, we also tested on dataset with motions but the camera drift anyway. 

Concerning the parameter, we kept the default value in file rovio.info. And as you mentioned, our accleromet is about 10 m/s^2, maybe it causes the problem.  

Thanks agin for your reply.

ayumizll, 2016-01-25:
Also I want to know does the algo ROVIO need for a accelerometer with high precision. 
Our IMU mpu6050
[image]
Do we need to change this IMU module...

andre-nguyen, 2016-01-25:
I was able to get rovio working with an mpu6000 so no I don't think you **need** an industrial quality IMU as long as you get your configuration right.

ank700, 2016-05-20:
Hi @ayumizll, were you able to work it out with your hardware setup as you said above. I am using UI 3060 with 12mm lens and mpu 6050, and having the same drift problem. 

In my case, initially when the camera+imu unit is stationary, everything is fine, the orientation, pose and twist is almost constant. When I move the camera slightly, all values start increasing/decreasing continuously (they are still changing even if I stop the movement and place it on a stable surface) .

This same thing is happening when I ran the machine hall 01 dataset with the rovio.config and euroc_cam0.yaml files given with the src code.
What could be wrong? I mean, when running live odometry, my config.info could be wrong but what about the dataset.

Thanks

bloesch, 2016-05-23:
@ank700, not to sure what could be wrong. I'll check the machine hall 01. Does it work on the vicon datasets?

Also, double check your camera-imu extrinsics. And how did you synchronize and timestamp your msgs?

andre-nguyen, 2016-05-23:
I tried the machine hall dataset a while ago and it was fine. Tracked the whole thing all the way.

ank700, 2016-05-24:
To run the machine hall dataset, I am doing the following steps:
1) i downloaded the calibration dataset 
2) calibrate the camera system as follows:
    kalibr_calibrate_cameras --bag cam_april.bag --topics /cam0/image_raw /cam1/image_raw --models pinhole-radtan pinhole-radtan --target april_6x6.yaml

This generates camchain-cam_april.yaml from which I create 2 cam.yaml files for each camera as follows:
 
image_width: 752
image_height: 480
camera_name: cam0
camera_matrix:
  rows: 3
  cols: 3
  data: [459.16206077071286, 0, 366.6065169928482, 0, 457.92646769906503, 248.44363379349446, 0, 0, 1]
distortion_model: radial-tangential
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.2890981920735853, 0.07947124044540965, 0.0002969539947434495, -3.3592673309414856e-06]

and similarly for cam1 in the same format as euroc_cam1.yaml.

3) Now, I launch the rosbag launch file. Using rovio.info

It gives 2 errors- ERROR: no camera Model detected!Testing Prediction
                           ERROR: No valid coordinate data!

The yaml files are the same as euroc_cam0.yaml but do not work. Also, after a few seconds , all the features shown in the visualizer are represented by blue numbers which are again continuously increasing.

bloesch, 2016-05-24:
you could just use the yaml files that are already provided (cfg/euroc_cam0.yaml). if you want to use your own calibration you should write "plumb_bob" instead of "radial-tangential" (kalibr is not fully compatible with rovio).

ank700, 2016-05-24:
I ran rovio_rosbag launch file with euroc..yaml file, and I get the following output. I have recorded the screen and am sharing the dropbox link. You will have to download to see it in good quality.

https://www.dropbox.com/s/51klld5mfpmp36d/Screencast%202016-05-24%2012%3A02%3A10.mp4?dl=0

bloesch, 2016-05-24:
in your video rovio says it uses the camera matrix with focal lengths around 800. if you use the default yaml file that is provided with rovio you should get:
```
Set Camera Matrix to:
458.654       0 367.215
      0 457.296 248.375
      0       0       1
Set distortion parameters (Radtan) to: k1(-0.283408), k2(0.0739591), k3(0), p1(0.00019359), p2(1.76187e-05)
```

something is wrong with the yaml file you are loading. this can be specified either through rosparam (prioritized) or in the rovio.info file.

ank700, 2016-05-24:
Thanks @bloesch , i was making a very silly mistake. Sorry for wasting your time. The whole calibration stuff is a bit too tedious. Anyway, now the dataset is working with stereo configuration. 

Regarding synchronization, is it right to perform synchronization of imu and camera msgs using system time? I have not done it yet, but will do it today.

bloesch, 2016-05-24:
You can use the system time as approximation as long as you can garantee that there is no large time delay between when the measurements are made and when it arrives on your system.

Alternatively you can employ trigger lines to either trigger the measurement processes of the sensors (a slave/master system works as well) or to record the measurement times on an another system.

ank700, 2016-05-24:
@bloesch I will try first with system time synchronization using http://wiki.ros.org/message_filters
If this does not give enough data pairs then I will have to look into hardware/software triggering.

bloesch, 2016-05-24:
just set the ros timestamps to the current time once your messages arrive on your system. you do NOT need to pair them up (and you will loose a lot of IMU msgs if you do). rovio will handle this internally. the only thing that is important is that the timestamps are expressed w.r.t. a common clock.

closed 2016-06-03 by ayumizll

juwikuang, 2017-05-05:
@ank700 what mistake you made? I have the same problem and I do not know what mistake I made. Thanks.