ethz-asl/rovio #38: How to set parameters of rovio with the result of kalibr

francescoli, 2016-02-15:
Hi,
  I've managed to get a acceptable calibration result of my smartphone with kalibr. But I'm confused with setting the parameters of rovio in the info file. As mentioned in #25 , some parameters are critical.
For example, qCM, I calculate it from the transformation matrix T_ic, am I right? Any other important parameters should be taken into consideration?

Regards

bloesch, 2016-02-15:
the extrinsics (qCM and MrMC) should be directly read from the T_ic. If I am not wrong MrMC should directly match the last column of T_ic and qCM should be derived from the rotation matrix (you might have to invert it).

Then you should be able to read the intrinsics from the calibr output as well. This include the camera matrix (4 parameters) and the distortion parameter. please check that you properly specify plumb_bob (= radial tangential) or equidistant in the yaml file. you can have a look at euroc_cam0.yaml for a template.

Then you should probably adapt the values of the PredictionNoise as well. Since you typically have a cheap IMU on smartphone you should increase the values of vel, acb, gyb, and att. Try to increase them separetely by factors of 10 until you have nice results. avoid to large values on gyb and acb.

A last things that come to my mind is that you might want to decrease the computational load of the algorithm. For this you can reduce the number of features to 15 (recompilation required) and increase trackingUpperBound and trackingLowerBound to 0.999 (so you have more tracked features). Also you can reduce the patch size to 4 (recompilation required).

andre-nguyen, 2016-02-15:
Just to pitch in, T_ic (and T_ci) are essentially homogenous 3d transform matrices.

http://planning.cs.uiuc.edu/node104.html

burrimi, 2016-02-15:
I added at some point scripts to generate the calibration file for rovio. It's in 
https://github.com/ethz-asl/kalibr/pull/35
I'll merge as soon as I have time to finish that PR.

francescoli, 2016-02-16:
@bloesch you mean transfer the inverse of the rotation block of T_ic to quaternion. Umm, seems I've made a mistake.

bloesch, 2016-02-16:
You can also just invert the obtained quaternion (change the sign of the x, y, and z components). However, this depends on the type of convertion you are using (depending on conventions), so its always best to try both, the non-inverted and the inverted form (one is definitely not gonna work). The easiest is probably if you checkout @burrimi 's scripts since they've been used on our MAVs.

francescoli, 2016-02-16:
Yes, thank you all above

closed 2016-02-16 by francescoli

arashazimi0032, 2020-09-22:
hi 
I want to change the parameters of the ROVIO, such as the ROVIO_NMAXFEATURE, and I do this in file rovio_node.cpp, but there is no change in how the algorithm runs . What is the problem?
Do I have to rebuild?
Thanks

rikba, 2020-09-24:
Yes @arashazimi0032 

If you change values of the parameters in the CMakeLists.txt, such as, [ROVIO_NMAXFEATURE](https://github.com/ethz-asl/rovio/blob/master/CMakeLists.txt#L4), you need to rebuild. Maybe you have to enforce cmake, i.e.,

```catkin build --this --force-cmake```

arashazimi0032, 2020-09-24:
> Yes @arashazimi0032
> 
> If you change values of the parameters in the CMakeLists.txt, such as, [ROVIO_NMAXFEATURE](https://github.com/ethz-asl/rovio/blob/master/CMakeLists.txt#L4), you need to rebuild. Maybe you have to enforce cmake, i.e.,
> 
> `catkin build --this --force-cmake`

Thanks a lot