ethz-asl/rovio #41: Reset/Init Rovio with initial external pose

mhkabir, 2016-03-01:
What would be the best way to implement implement reseting/initialising Rovio with a pose estimate from an external filter? 

I want to init Rovio in the beginning with an initial pose, and then if it ever loses track/diverges, reset Rovio to the last know good pose (again from an external topic). 

What is the way to do this best? Including detection of divergence?

bloesch, 2016-03-02:
resetting rovio with an externally provided pose should be relatively simple. For now there is a resetWithAcceleromter (rovioFilter) function which uses an accelerometer measurement to initialize the inclination angles, while the rest of the state is initialized by copying the init_ state. So the simplest way to get such a reset functionality would be to implement a ros service which would basically set the init_ state to the desired pose and set isInitialized_ (rovioNode) to false which will trigger a re-initialization.

you also mentioned the detection of divergence, and I think this might be a bit more involved. there are many different aspect that hint at divergence and you might want to combine them to get a robust divergence detection. possible criteria are:
- low number of tracked features
- high number of new feature initialization
- increasing velocity estimates (and covariance)
- large feature distances
some of those criteria might however also be observed when you have a high scene distance, e.g. if you fly high above the ground with a MAV...

mhkabir, 2016-03-05:
@bloesch Thanks!
I added a external pose subscriber to the node, and added a service call. The idea is to set the initial state to whatever is sent on the topic, and just reset to that when the service is called.

Sounds good?

bloesch, 2016-03-07:
what about passing the desired initial pose as service parameter? would avoid having an extra topic

mhkabir, 2016-03-09:
@bloesch I didn't do that because I wanted to have it such that we could initialise directly from a measurement on that topic (even for the first init). 

Is architechturally exactly what @raghavkhanna has implemented here : https://github.com/ethz-asl/rovio/tree/feature/init_yaw

Just that mine adds a full pose estimate, and an auxillary service call. It might be worth getting the init_yaw patch and mine together for completeness. :)

bloesch, 2016-03-10:
I still feel like passing the desired initial pose as parameter of the service would be the more generic way. But I understand that depending on your ros setup it might be more convenient to read it from topic. @raghavkhanna what is your opinion on that?

raghavkhanna, 2016-03-10:
Agree with @bloesch in general. I made init_yaw primarily for working with GPS outdoors, since we have a global yaw from the magnetometer and would like to align the Rovio world frame with a global ENU frame. Since I don't get complete pose measurements, I would leave that function in there. I noticed that there's already a initWithIMUPose function, so couldn't we just use that :) for what @mhkabir wants?

mhkabir, 2016-03-11:
> BTW, @bloesch <https://github.com/bloesch>, we tested Rovio + GPS
> outdoors for waypoint navigation flying today. The global drift issue seems
> to be resolved :). We didn't observe overall drift > 1 m for a 50 m or so
> trajectory.

@bloesch @raghavkhanna We're good then. This is exactly what I've done. Let
me clean it up, and give to you a PR . I used initWithImuPose in my
implementation.

Raghav, some questions -

How are you projecting the global coordinates to a local frame? Are you
using a separate node? Can you link me to your config files or post them
here?

Regarding the mag transformation - How are you converting the measurements
into an IMU message?

closed 2016-06-02 by schneith