ethz-asl/rovio #47: Better feature initialization

devbharat, 2016-03-12:
I was wondering if it makes sense to initialize(or look for) new features in specific 'sections' of the image based on current pose uncertainty of the camera to gain most information with least amount of new features, rather than spreading many features everywhere. Surely, based on current pose uncertainty of the camera and locations currently tracked features, new features in certain directions will yeild more information to better localize the camera than other directions ? @bloesch

bloesch, 2016-03-15:
I completely agree. In the current state the detection of new features is in a terrible state and very inneficient. Depending on the resolution and the detection rate it is very costly and poses more computational costs than all the rest together.

For now there is a simple bucketing scheme which doesn't work too badly. But I think initializing features which provide "better" information would definitely be beneficial. While this can include the location of the feature, it could also considere the texture of the corresponding patch.

If somebody has some ideas and is willing to implement this I would be glad to help.

Kushine, 2018-01-24:
How to get dataset trajectory the method generated?Have you got it? @devbharat