ethz-asl/rovio #49: Evaluation tool

bloesch, 2016-03-30:
There is a python based evaluation tool for comparing rovio's output with other VIO and groundtruth ([https://github.com/ethz-asl/trajectory_toolkit](https://github.com/ethz-asl/trajectory_toolkit)). It also provides different plotting functionality. Unfortunately it has not been updated for quite a while.

Denfender, 2016-10-23:
hallo,i do not  know how to use this tool,could you tell me how to use this in details.thanks a lot @bloesch

bloesch, 2016-10-25:
it depends what you want to do.... this tool is mainly usefull if you want to compare with groundtruth.

Denfender, 2016-10-26:
thank for your reply! i want ro compare with groundtruth ,but i do not know the mean of "rovioEvaluator.pyth"in rovio_evaluation.py .and how can i use it to show the trajectory of VIO and groundturth when i run the datesets. thanks a lot .@ bloesch

bloesch, 2016-10-26:
you need to adapt the bag/gtFile as well as the odomTopic/gtTopic in rovioEvaluator. then try to run the script step by step.

please post future questions in https://github.com/ethz-asl/trajectory_toolkit