ethz-asl/rovio #53: Problems compiling the code in Ubuntu 14.04

ank700, 2016-05-11:


ank700, 2016-05-11:
Hello, I installed rovio on my pc. Config. - Ubuntu 14.04 ROS Indigo

I made the changes required in the yaml file and info file. I also changed the launch file since I am using only one camera (rgb). 
When I run the rovio_node.launch file, a screen appears showing the axes at the screen's centre and in the terminal window it says - 

Set Camera Matrix to:
874.798       0  329.03
      0 873.424 259.539
      0       0       1
Set distortion parameters (Radtan) to: k1(0.001671), k2(0.003223), k3(0), p1(0.001963), p2(0.002147)
Testing Prediction
==== Test successful (9.38621e-08) ====
==== Test successful (7.69526e-08) ====
Testing cameraOutputCF
==== Test successful (6.10703e-08) ====
Testing imuOutputCF
==== Test successful (1.22913e-08) ====
Testing attitudeToYprCF
==== Test successful (3.62656e-08) ====
Testing transformFeatureOutputCT
==== Test successful (6.07747e-09) ====
Testing LandmarkOutputImuCT
==== Test successful (2.19615e-08) ====
Testing FeatureOutputReadableCT
==== Test successful (1.10557e-08) ====
Testing pixelOutputCT (can sometimes exhibit large absolut errors due to the float precision)
==== Model jacInput Test failed: 7.52085 is larger than 1 at row 1(def_0) and col 0() ====
  14441.1  14433.6
Testing zero velocity update
==== Test successful (3.04311e-10) ====
==== Test successful (1.39778e-10) ====

Can someone help me setting up rovio correctly. It will be a lot of help.
The hardware used is - normal logitech hd cam with rolling shutter and auto focus
                                    - crius imu 
Thanks

mhkabir, 2016-05-12:
Looks like IMU messages never arrive. Filter doesn't initialise. 

Have a look at your remappings, and check if you're providing IMU data to the correct subscriber.

ank700, 2016-05-18:
Hello @mhkabir , I am sorry for replying so late.
I changed the rovio_node.launch file as follows:

<launch>
  <node pkg="rovio" type="rovio_node" name="rovio" output="screen">
  <param name="filter_config" value="$(find rovio)/cfg/rovio.info"/>
  <param name="cam0_config" value="$(find rovio)/cfg/uicam.yaml"/>
  <param name="imu0_topic_name" value="imu/raw_data"/>
  <param name="cam0_topic_name" value="/camera/image_raw"/>
  </node>
</launch>

Now on running this launch file I get in the terminal:
roslaunch rovio rovio_node.launch 
... logging to /home/ros/.ros/log/60a4b238-1d10-11e6-a003-3010b3a20721/roslaunch-ros-4680.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ros:53805/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /rovio/cam0_config: /home/ros/rovio_w...
 * /rovio/cam0_topic_name: /camera/image_raw
 * /rovio/filter_config: /home/ros/rovio_w...
 * /rovio/imu0_topic_name: imu/raw_data

NODES
  /
    rovio (rovio/rovio_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rovio-1]: started with pid [4760]
Testing Prediction
==== Test successful (9.38621e-08) ====
==== Test successful (7.69526e-08) ====
Testing cameraOutputCF
==== Test successful (6.10703e-08) ====
Testing imuOutputCF
==== Test successful (1.22913e-08) ====
Testing attitudeToYprCF
==== Test successful (3.62656e-08) ====
Testing transformFeatureOutputCT
==== Test successful (6.07747e-09) ====
Testing LandmarkOutputImuCT
==== Test successful (2.19615e-08) ====
Testing FeatureOutputReadableCT
==== Test successful (1.10557e-08) ====
Testing pixelOutputCT (can sometimes exhibit large absolut errors due to the float precision)
==== Test successful (0.000992549) ====
Testing zero velocity update
==== Test successful (3.04311e-10) ====
==== Test successful (1.39778e-10) ====

Even with all the tests successful, it does not show anything on the scene window, apart from a grid and coordinates at the centre.

The list of subscribers is 
rostopic  list -s
/camera/image_raw
/camera/ueye_manager/bond
/image_raw
/imu/raw_data
/rosout

I think the filter is still not initialising. What should I check? I am trying to understand the code, but have not found anything yet.

bloesch, 2016-05-18:
Maybe use some tool like rqt_graph to double check if all topics are properly hooked up. Also, you might miss a backslash in "/rovio/imu0_topic_name: imu/raw_data".

ank700, 2016-05-19:
Thank you @mhkabir @bloesch , it is now working. The problem was both the topics /camera/image_raw and /imu/data were not subscribed by rovionode. I changed the topic names in the rovionode file. Then, my Imu data was published as a custom message, so I had to convert it to sensor_msgs/Imu.

Now I can experiment and set correctly the other parameters.

closed 2016-05-20 by ank700