ethz-asl/rovio #54: Compatibility issue with ubuntu 16.04

cynosure4sure, 2016-05-24:
Hi! I tried compiling ROVIO on ubuntu 16.04 with ROS jade, got over 100 build errors another person compiled older version of boost,opencv etc to see if he can resolve such errors with no luck. Anyone else facing similar issues?

bloesch, 2016-05-25:
I have not tried compiling on 16.04 yet. What are the errors mostly related to? ros? opencv? boost?

supermice, 2016-07-12:
I successfully compiled rovio on ubuntu 1604 with ROS kinetic. Unfortunately, when I run the launch file(I have already edited the data set path), the process has died. Do you guys have any idea about this?

Here are the outputs
`zerotech@zerotec:~/git$ roslaunch rovio rovio_rosbag_node_wfhedit.launch 
... logging to /home/zerotech/.ros/log/b2bd7a96-4803-11e6-80b4-3c46d8c36435/roslaunch-zerotec-9089.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zerotec:38490/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.2
 * /rovio/cam0_topic_name: /cam0/image_raw
 * /rovio/cam1_topic_name: /cam1/image_raw
 * /rovio/camera0_config: /home/zerotech/gi...
 * /rovio/camera1_config: /home/zerotech/gi...
 * /rovio/filter_config: /home/zerotech/gi...
 * /rovio/imu_topic_name: /imu0
 * /rovio/rosbag_filename: /home/zerotech/ro...

NODES
  /
    rovio (rovio/rovio_rosbag_loader)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rovio-1]: started with pid [9107]
[rovio-1] process has died [pid 9107, exit code -11, cmd /home/zerotech/git/rovio_ros_ws/devel/lib/rovio/rovio_rosbag_loader __name:=rovio __log:=/home/zerotech/.ros/log/b2bd7a96-4803-11e6-80b4-3c46d8c36435/rovio-1.log].
log file: /home/zerotech/.ros/log/b2bd7a96-4803-11e6-80b4-3c46d8c36435/rovio-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

bloesch, 2016-07-28:
sorry for the delayed reply. have you tried running the regular launch file (rovio_node.launch)? otherwise you might want to try compiling in debug mode and run it through gdb.

ke-sun, 2016-08-27:
I have tried the master branch of rovio with master branch of kindr on 16.04 with ROS Kinetic. Here is the output of gdb

```
Starting program: /home/ke/RosWorkSpace/simulator/devel/lib/rovio/rovio_rosbag_loader __name:=rovio __log:=/home/ke/.ros/log/2cfdeab4-6bf6-11e6-b1ca-44850039ad73/rovio-1.log
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe61d7700 (LWP 15320)]
[New Thread 0x7fffe59d6700 (LWP 15321)]
[New Thread 0x7fffe51d5700 (LWP 15322)]
[New Thread 0x7fffe49d4700 (LWP 15327)]

Thread 1 "rovio_rosbag_lo" received signal SIGSEGV, Segmentation fault.
0x00000000005599c5 in rovio::State<25u, 4, 6, 1, 0>::State() ()
```
And here is the output of `bt`,

```
#0  0x00000000005599c5 in rovio::State<25u, 4, 6, 1, 0>::State() ()
#1  0x0000000000562ffd in LWF::FilterState<rovio::State<25u, 4, 6, 1, 0>, rovio::PredictionMeas, rovio::PredictionNoise<rovio::State<25u, 4, 6, 1, 0> >, 0u>::FilterState() ()
#2  0x0000000000563b98 in rovio::FilterState<25u, 4, 6, 1, 0>::FilterState() ()
#3  0x0000000000564797 in LWF::FilterBase<rovio::ImuPrediction<rovio::FilterState<25u, 4, 6, 1, 0> >, rovio::ImgUpdate<rovio::FilterState<25u, 4, 6, 1, 0> >, rovio::PoseUpdate<rovio::FilterState<25u, 4, 6, 1, 0>, -1, -1> >::FilterBase() ()
#4  0x000000000056593c in rovio::RovioFilter<rovio::FilterState<25u, 4, 6, 1, 0> >::RovioFilter() ()
#5  0x000000000047b8c0 in main ()
```

bloesch, 2016-08-27:
A fix for issues related to 16.04 is proposed in #83.

closed 2016-08-27 by bloesch