ethz-asl/rovio #55: can rovio be adapted to used rgb-d camera?

learnsor, 2016-05-31:
as I am new to rovio, I wonder if rovio can be used with rgb-d camera ?

bloesch, 2016-05-31:
This could probably be done. You could add depth measurements for every landmark. If properly done this could increase the robustness of the system - I doubt that you would gain a lot in accuracy.

In terms of implementation, adding depth measurements is not too difficult. The measurement and update class could be relatively easily extended. The interfacing (ros node) could be a bit more cumbersome, especially if you want to keep compatibility with RGB-only.

yongbos, 2018-03-07:
Thank you for awesome algorithm.
I want to adapt rgb-d information to ROVIO instead of stereo camera.
As you mentioned "in terms of implementation", where can I add depth measurements?
Would you explain more detail? or example of adding depth measurements.
I look forward to your reply.
Thank you.

bloesch, 2018-03-09:
Adding depth measurements would require major changes to rovio, but could improve its performance. If I am right, the code you mention above is only for initialisation of the features. If you really want the depth information to be used you will have to add a proper measurement: for every feature which is in the current frame you could add a depth measurement model like: dept_meas = depth_state + noise (with proper transformation and units).