ethz-asl/rovio #60: Scene window is still

wf34, 2016-06-02:
The scene window is still and shows nothing but the plane grid and the camera axis at the origin.
Everything else is workable, the node publishes odometry updates, the scene calls the render function.
Would you like to share cues on how to investigate that?
I run rovio on Ubuntu 14.04 with OpenCV 3.0.0.

bloesch, 2016-06-03:
I never tested the implementation with OpenCV3, but I think others have. Does the output of the odometry look reasonable, i.e., is it within a couple of meters range?

wf34, 2016-06-03:
Yes, it seems to be correct on the Machine Hall 01.

bloesch, 2016-06-03:
this is weird. does the image visualization look alright? is the tracking stable (features are mainly green)?

wf34, 2016-06-04:
Yes, the tracking is of the same quality as in the demonstration video.
I also had rovio rebuilt with OpenCV 2.4.13, the result is the same.
It is probably worth mentioning, that I have built cv_bridge from sources too, as the packaged by ubuntu version of opencv, which was pulled as cv_bridge dependency, resulted in linking errors in rovio for me.

bloesch, 2016-06-06:
I doubt that this has to do something with opencv or the cv_bridge. more lickely some graphics issues. you could try checking/printing the values in RovioScene/Scene.

alternatively you could try using rviz for visualization.

closed 2016-09-19 by bloesch

YonatanSimson, 2021-07-05:
I had the same issue and found out what the issue was. If you try running imshow in a thread it can either crash or freeze. 

For me the solution was to declare the windows using namedWindow before using imshow