ethz-asl/rovio #61: Any reason for the virtual calls in lightweight_filtering?

nicolov, 2016-06-05:
It seems that the lightweight_filtering classes (for example in States.hpp) have all the machinery in place to avoid virtual function calls using the [CRTP pattern](https://en.wikipedia.org/wiki/Curiously_recurring_template_pattern). Is there any reason to keep the virtual calls instead of using the `static_cast` trick to avoid the runtime overhead?

Thanks

bloesch, 2016-06-06:
I guess you could get rid of the virtual function calls. I am currently thinking of refactoring lightweight_filtering, but I will probably try to get rid of many of the templated schemes. The motivation is to reduce compilation time and improve code readibility. I doubt that this will notably decrease performance since the computational costs are mainly given by the matrix multiplications in Eigen.

nicolov, 2016-06-06:
Sure, just wondering why. It would be a pity to loose all the neat composable states in LWF::State tough. But you're right that the prediction/update API can drop templates with little loss.

bloesch, 2016-06-06:
The composable states are defintely rather convenient. I am not too sure how I could achieve this without the templates.

Concerning your original question: there were multiple iterations on the State class (the CRTP came in later), and as this was my first more intensive use of templates not all implementations follow best practice =)

nicolov, 2016-06-06:
On the other hand, you rarely (if ever) operate on multiple state elements at the same time (i.e. boxPlus over multiple features) that would warrant the composable state types. Considering this, why have the super-complex type definition for the FilterState, and then multiple named functions to access the individual elements? The interface is no different than a struct or class with accessors, right?

I guess you could keep the same genericity by templating the FilterState, predictions, and updates over the few compile-time constants that change its components (basically the extrinsic calibration), and stopping the template magic there.

bloesch, 2016-06-06:
The filter has to operate on the full state (e.g. Kalman update) and keep track of the joint covariance. The lightweight filtering (LWF) library is also used in other state estimation framework which may have completely different state/measurement structures. Of course, you could write a dedicated Kalman filter for rovio and this could be kept very minimalistic. Still, there are some tools in LWF which are rather convenient (e.g. finite difference Jacobian testing, or the easy support for 3D rotations).

In a general Kalman filter library, the design of the state class is a central part. It should include the following functionality:
- Composable from multiple substates (e.g. scalars, vectors, quaternions)
- The composition should directly define boxminus/boxplus and a couple of utility functions of the full state
- Allow easy access to the substates (ideally typesafe).

I guess this could be done in a template-free manner. Given a certain state size it is not really important anymore to know the dimensions at compile time.

nicolov, 2016-07-01:
Thanks for the useful info. I've spent some time practicing with overkill template design though :) Let's say you have a few kind of states (manifolds that implement boxPlus, etc..)

```cpp
template <class Class, int n>
struct Manifold {
    static constexpr int dim = n;
};

struct Quaternion : Manifold <Quaternion, 3>{
    void print() const { std::cout << "A quaternion\n"; }
};

struct Vector3 : Manifold <Vector3, 3> {
    void print() const { std::cout << "A vector3\n"; }
};

struct Normal : Manifold <Normal, 2> {
    void print() const { std::cout << "A normal\n"; }
};
```

Following the approach [here](http://duriansoftware.com/joe/index.html), you define a struct to hold the shape of the state and the name of the states:

```cpp
struct StateDef {
    struct WrWM {
        using type = Vector3;
        type WrWM;
        constexpr static char const * name() { return "WrWM"; }
        type &value() & { return this->WrWM; }
        type const &value() const & { return this->WrWM; }
    };

    struct n {
        using type = Normal;
        type n;
        constexpr static char const * name() { return "n"; }
        type &value() & { return this->n; }
        type const &value() const & { return this->n; }
    };
};
```

And finally, you build the complete state type by multiple inheritance over the state components:

```cpp
template<typename...Fields>
struct Struct : Fields... {
    // Preserve default constructors
    Struct() = default;
    Struct(Struct const &) = default;

    using as_tuple = std::tuple<Fields...>;

    void print() {
        // http://stackoverflow.com/questions/7230621
        int x[] = {(this->Fields::value().print(), 0)...};
        (void)x;
    }

    constexpr size_t dim_f() {
        // Build an array of element dimensions using pack expansion
        size_t x[] = {Fields::type::dim...};
        size_t c = 0;
        for (size_t i = 0; i < std::tuple_size<as_tuple>::value; i++) {
            c += x[i];
        }

        return c;
    }
};
```

The cool thing is that you can access the state elements by name: 

```cpp
s.WrWM.print();
```

and use either `constexpr` functions at compile time (like in `dim_f()`) or parameter pack expansion (like in `print()`)at runtime to operate on the whole state at the same time:

```cpp
using State = Struct<StateDef::WrWM, StateDef::n>;

int main() {
    State s;
    static_assert(s.dim_f() == 5, "");
    std::cout << "Operating over all elements\n";
    s.print();
    std::cout << "Operating on a single element:\n";
    s.WrWM.print();
}
```

Yes, I realize this is useless, but maybe could give someone ideas for a next version or project.

bloesch, 2016-07-01:
I think the solution is interesting - not to familiar with the expansions though. In the case of boxplus you would have to split a vector into smaller vectors which match the dimensions of the components. But I guess it should be feasible to pass arguments to your "unpacked methods"?

I definetly like the direct access to the components. This is way nicer than the current templates get<>() stuff. I guess I'll investigate this approach before deciding to get rid of the templates or not.

One question: "StateDef" is only used as namespace, is it?

nicolov, 2016-07-04:
Yep, `StateDef` is just a namespace. On the page I linked there's also a simple preprocessor macro to shorten state element definitions to one line.

Worst case, you can always use recursive templates as you're doing now, but constexpr functions are supposed to be easier on the compiler.

nicolov, 2016-08-03:
After testing on a slow Atom processor (compile time was about 30 minutes..), I've come up with an even weirder idea, using a python code generator called [cog](https://www.python.org/about/success/cog/).

The states are defined in Python code embedded in the C++ source, like this:

```python
class MyState : public MyStateBase<MyState> {
public:
  /* [[[cog
  import lwf_cog
  state_def = (
    ("sca", "LWF::ScalarElement"),
    ("vec0", "LWF::VectorElement<3>"),
    ("vec1", "LWF::VectorElement<3>"),
    ("vec2", "LWF::VectorElement<3>"),
    ("vec3", "LWF::VectorElement<3>"),
    ("qua0", "LWF::ArrayElement<LWF::QuaternionElement,2>"),
    ("qua1", "LWF::ArrayElement<LWF::QuaternionElement,2>"),
    # ("aux", "AuxillaryElement"),
  )
  state_name = "MyState"
  lwf_cog.define_state(state_def, state_name)
  lwf_cog.define_dim(state_def, state_name)
  lwf_cog.define_boxplus(state_def, state_name)
  lwf_cog.define_boxminus(state_def, state_name)
  lwf_cog.define_boxminusjac(state_def, state_name)
  lwf_cog.define_setidentity(state_def, state_name)
  ]]]*/
  // [[[end]]]
};
```

and Python functions like this one print out the appropriate C++ code:

```python
def define_boxplus(state_def, state_name):
    cog.outl("void boxPlus(const mtDifVec& vecIn, %s& stateOut) const {" % state_name)
    for i, (el_name, el_type) in enumerate(state_def):
        cog.outl("""
  {{
    constexpr unsigned int cum_dim = {cum_dim};
    {n}.boxPlus(vecIn.template block<{T}::D_,1>(cum_dim,0), stateOut.{n});
  }}
""".format(T=el_type,
           n=el_name,
           cum_dim=" + ".join(["0"] + ["{}::D_".format(el_type) for _, el_type in state_def[:i]])))
    cog.outl("}")
```

It's a bit of an hack but the generated code doesn't look bad and is as fast as the templates (since most functions can be `constexpr` anyway). CMake can handle the generation step:

```cmake
ADD_CUSTOM_COMMAND(OUTPUT ${CMAKE_SOURCE_DIR}/src/testState.cpp
                   COMMAND PYTHONPATH=${CMAKE_SOURCE_DIR} cog.py ${CMAKE_SOURCE_DIR}/src/testState.cpp.cog > ${CMAKE_SOURCE_DIR}/src/testState.cpp
                   DEPENDS ${CMAKE_SOURCE_DIR}/src/testState.cpp.cog)
```

bloesch, 2016-08-05:
I have thought about code generation and there are a couple of approaches in this direction (e.g. Nasa's AUTOFILTER). I like your python approach (allthough the code printing looks a bit messy =) ). In the end the user only has to specify the state_def. But it probably gets more difficult to use if adaptations/extensions are required.

nicolov, 2016-08-05:
Cool, I didn't know about the nasa project, it seems even more higher level.

I agree the generator here is not production quality. One alternative would
be using one of the python packages to generate an abstract syntax tree.
Another would be a proper template language, like ninja (used in
Python-based websites).

Couldn't extension be handled through inheritance? Extending the templated
code is not easy, either. Templight or similar template debugging tools all
choke during compilation.

bloesch, 2016-08-05:
I guess code generation could simplify a couple of things and probably also speed up compilation time (in comparison to the current templating). But I prefer staying with c++ only where I feel I can better estimate the expected workload.

closed 2016-09-19 by bloesch

nicolov, 2017-05-29:
I ended up actually trying code generation just to see if it would work, and it did. That's how a state definition looks like:

```
namespace LWFTest {

{{ m.def_state_class("TestState", [
    ["LWF::ScalarElement", "sca"],
    ["LWF::VectorElement<3>", "vec0"],
    ["LWF::ArrayElement<LWF::QuaternionElement, 2>", "qua0"],
]) }}

}
```

The Python/Jinja2 "library" code is [here](https://github.com/nicolov/rovio_fork/blob/master/lightweight_filtering/templates/_macros_library.j2). The messiest part is the special case for handling of arrays (nested arrays are not supported). In any case, it worked well enough to support rovio.

bloesch, 2017-06-12:
looks interesting! I am not too familiar with code generation, but I guess that by using code generation one could get rid of many of the complexities in the LWF.