ethz-asl/rovio #63: rovio convention for camera coordinate system

berndpfrommer, 2016-06-09:
I'm trying to get rovio to run on video/imu data recorded on a Google Tango (handheld device, using RGB camera). It's working on the Euroc data set, but not on the Tango data. I believe the camera calibration is o.k., but I'm not sure I got the IMU-to-camera transformation right.

What convention are you using for the camera coordinate system? The standard one (x to the right , y down, z along viewing direction)?

When looking at the imu data:

linear_acceleration: 
  x: -0.057368904
  y: 10.583193
  z: 3.0418723

it appears that y is pointing up.
Is there a way to print out what the gravitational vector "g" looks like in camera coordinates? Basically any method that assures that the transform is correct?
[Here is a video of rovio in action](https://youtu.be/teguQ_r1CuMl), but not quite working yet.

Any help is appreciated!

maciejmatuszak, 2016-06-09:
@berndpfrommer I think I just went trough very similar exercise as you. Have a look at #62.
Basically those the the things that got me:
- the camera coordinate systems z axes is along the optics axes (I missed that but it may be obvious to someone that works with camera xyz coordinates more than once)
-  to gain confidence in the rotation you may want to run the Kalibr dynamic calibration [Camera IMU calibration](https://github.com/ethz-asl/kalibr/wiki/camera-imu-calibration) it outputs out the rotation matrix and then you can run the conversion too lto get rovio config. This is How I run it first static:
[kalibr_static.sh.txt](https://github.com/ethz-asl/rovio/files/308080/kalibr_static.sh.txt) then dynamic 
[kalibr_dynamic.sh.txt](https://github.com/ethz-asl/rovio/files/308081/kalibr_dynamic.sh.txt) To tell you the truth I only used the quaternion that it prints. I did the distortion calibration using the ros camera_calibration it gives you back plumb-bob what is needed for rovio.
- last and most important the latest version require some tuning at least for my setup, looks like I have big delay between imu and images data. This commit: 02753db02854549c48fc3a5850a60b29feaf3dd1 works out of the box even with my crappy data :) It is my understanding this is very close to the version the ethz team publish paper with.
@bloesch I hope I did not misrepresented anything, just trying to help.

bloesch, 2016-06-10:
Thanks @maciejmatuszak for the help.

The camera z-axis is indeed along the optical axis, x right, y down.

For some reasons a previous commit tends to be slightly more robust. In the end I would still recommend to use the newest one together with the adapted parameters.

You can look at the virtual horizon in the camera frame as a first sanity check if your imu-camera orientation is right (this might not work if rovio drifts due to other reasons). If the real horizon matches the virtual horizon there is only a remaining yaw ambuigity.

berndpfrommer, 2016-06-10:
Thanks everybody for the fast reply. Unfortunately I no longer have access to said Tango device, so I have to live off of the recorded sequences. I also have some visensor sequences, but alas, the visensor seems to have died on me.
I will try to do the calibration with the tango, and revert back to the older commit to see if that helps.

berndpfrommer, 2016-06-13:
Some progress: got the vi-sensor to work again, so I should be able to calibrate the extrinsics, but I'm not sure I really have to, because after reverting to commit 02753db, things started to work much better. The virtual horizon stays level, and the camera intrinsic rotations that are written in "verbose" mode are almost identical to the (hopefully correct) starting guess, except that the translations are non-zero now (but just a few cm, exactly what's expected).
The other thing I noticed: the printout on the console is like this:
Camera extrinsics: 0
  0.999725446526 -0.0020157314521 -0.0223376928719 -0.0067820256182
  0.00632872525664 -0.0289046088724 0.0413183999772

Whereas my config file has:

Camera0
{
	CalibrationFile ;	
	qCM_x  -0.00140695072068;	
	qCM_y  0.00177042696561;
	qCM_z  0.00282151724388;
	qCM_w   0.999993462538;
}
So I assume that the quaternion printout is (w,x,y,z).

I noticed the code base of the old commit lacks some nice features (like creating an output bag), so I'd love to run on the new code base. I added code to allow running against a bag with compressed images and want to test that against the most recent code as well before I submit a PR.

Problem is that my diagnostic feedback reduces to something stupid like "it doesn't work". What should I look out for? I don't want to just fiddle randomly with parameters. Is there a systematic way to diagnose issues? I was hoping to e.g. first run completely without visual information, and get the pose just from the IMU to see that that works, then have some way of systematically determining if the optical part works, then maybe understand if the problem is with combining visual inertial.

bloesch, 2016-06-15:
The translational part of the camera-imu extrinsics will rarely be zero. Also you are correct about the order of the printed quaternion to be (w,x,y,z).

Concerning the diagnostics I am not too sure. It will be difficult to use one sensor modality only. But since you are using a visensor, and the code has been developed/tested with visensor data mainly, it should really work right out of the box. Unfortunately the camera calibration and the camera-imu extrinsics vary strongly between different visensors (they have mounted the IMU in different ways). But in the end you will just have to adapt those two calibration parameters.

Once the old commit works properly you should also be able to transfer those settings to the newer commit. You should probably adapt ImgUpdate.initCovFeature_0 to 0.5 (higher robustness against missing motion) and ImgUpdate.penaltyDistance to 100 (better distribution of features).

mhkabir, 2016-06-15:
@bloesch Just a comment - I never managed to get the new commit working in real time in any config (stereo or mono), when using the normal Rovio node. I believe that you've done most of your recent testing on bagfiles using the rosbag loader, and so haven't observed much of a problem. (referring to the end comments in https://github.com/ethz-asl/rovio/issues/40)

It's not only me, and multiple others (e.g @andre-nguyen) have reported that Rovio becomes unusable on the new commits due to the slowdowns(?). I think that the current master is way too computationally expensive (even with the pyramid levels and patch size reduced to the smallest feasible values) for application-level work.

bloesch, 2016-06-16:
@mhkabir thanks for the comment. I am aware of the computational issues related to the newest commit. I should find time over the summer to work on a new version.

closed 2016-09-19 by bloesch