ethz-asl/rovio #65: Workability on KITTI

wf34, 2016-06-17:
Hello everyone.
I was interested to find out how rovio would do on some universally known dataset. I tried to run it on a sequence from KITTI, yet got no sensible output there.
To sum up my endeavours:
* I followed wiki on configuration suggestions
* I also have implemented suggestions from issue #62, namely tried to checkout to January revision, suggested there (with none improvement).
* My setup works exactly as it is shown in the demonstration video on the author's Machine Hall dataset.
* I was tending to blame Camera-IMU calibration, so I even tried to pass exact calibration provided by KITTI.

 I'm going to attach KITTI sequence in form of [rosbag](https://yadi.sk/d/_JHL6ACpsM22T), and parameters I used. The screenshots attached - is the most workable version I managed to get. It is the third and fourth frames of dataset, around that moment  valid keypoints disappear to never appear again.
Thank you for you suggestions!

[image]
[image]

Launch file:
```
<?xml version="1.0" encoding="UTF-8"?> 
<launch>
  <node pkg="rovio" type="rovio_rosbag_loader" name="rovio_bag_reader_kitti" output="screen">
  <param name="filter_config" value="$(find rovio)/cfg/rovio_kitti.info"/>
  <param name="camera0_config" value="$(find rovio)/cfg/euroc_cam0_kitti.yaml"/>
  <param name="rosbag_filename" value="/home/XXX/datasets/rovio_bags/kitti_longer.bag"/>
  <param name="imu_topic_name" value="/imu0"/>
  <param name="cam0_topic_name" value="/cam0/image_raw"/>
  </node>
</launch>
```
Camera Calibration:
```
image_width: 1242
image_height: 375
camera_name: cam0
camera_matrix:
  rows: 3
  cols: 3
  data: [7.215377e+02, 0.000000e+00, 6.095593e+02, 0.000000e+00, 7.215377e+02, 1.728540e+02, 0.000000e+00, 0.000000e+00, 1.000000e+00]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data:  [0.0,  0.0, 0.0, 0.0, 0.0]
```

Rovio configuration:
```
Common
{
	doVECalibration false;		Should the camera-IMU extrinsics be calibrated online
	depthType 1;				Type of depth parametrization (0: normal, 1: inverse depth, 2: log, 3: hyperbolic)
	verbose false;				Is the verbose active
}
Camera0
{
     qCM_x 0.5010303840;
     qCM_y -0.4989766415;
     qCM_z 0.4982646186;
     qCM_w 0.5017202760;
     MrMC_x 0.8127456660;
     MrMC_y -0.2432397200;
     MrMC_z 1.0715037000;
}
Init
{
    State
    {
        pos_0 0;			X-entry of initial position (world to IMU in world) [m]
        pos_1 0;			Y-entry of initial position (world to IMU in world) [m]
        pos_2 0;			Z-entry of initial position (world to IMU in world) [m]
        vel_0 0;			X-entry of initial velocity (robocentric, IMU) [m/s]
        vel_1 0;			Y-entry of initial velocity (robocentric, IMU) [m/s]
[... 152 lines cut ...]
    pre_cov_MrMV 1e-4;							Covariance parameter of pose between inertial frames, linear part [m^2/s]
    pre_cov_qVM 1e-4;							Covariance parameter of pose between inertial frames, rotational part [rad^2/s]
    MahalanobisTh0 12.6511204;					Mahalonobis distance treshold of pose measurement
	doVisualization false;						Should the measured poses be vizualized
	enablePosition true;						Should the linear part be used during the update
	enableAttitude true;						Should the rotation part be used during the update (e.g. set to fals eif feeding GPS-measurement)
	noFeedbackToRovio true;						By default the pose update is only used for aligning coordinate frame. Set to false if ROVIO should benefit frome the posed measurements.
	doInertialAlignmentAtStart true;			Should the transformation between I and W be explicitly computed and set with the first pose measurement.
	timeOffset 0.0;								Time offset added to the pose measurement timestamps
    qVM_x 0;									X-entry of quaterion representing IMU to reference body coordinate frame of pose measurement (JPL)
    qVM_y 0;									Y-entry of quaterion representing IMU to reference body coordinate frame of pose measurement (JPL)
    qVM_z 0;									Z-entry of quaterion representing IMU to reference body coordinate frame of pose measurement (JPL)
    qVM_w 1;									W-entry of quaterion representing IMU to reference body coordinate frame of pose measurement (JPL)
    MrMV_x 0;									X-entry of vector representing IMU to reference body coordinate frame of pose measurement
    MrMV_y 0;									Y-entry of vector representing IMU to reference body coordinate frame of pose measurement
    MrMV_z 0;									Z-entry of vector representing IMU to reference body coordinate frame of pose measurement
    qWI_x 0;									X-entry of quaterion representing World to reference inertial coordinate frame of pose measurement (JPL)
    qWI_y 0;									Y-entry of quaterion representing World to reference inertial coordinate frame of pose measurement (JPL)
    qWI_z 0;									Z-entry of quaterion representing World to reference inertial coordinate frame of pose measurement (JPL)
    qWI_w 1;									W-entry of quaterion representing World to reference inertial coordinate frame of pose measurement (JPL)
    IrIW_x 0;									X-entry of vector representing World to reference inertial coordinate frame of pose measurement
    IrIW_y 0;									Y-entry of vector representing World to reference inertial coordinate frame of pose measurement
    IrIW_z 0;									Z-entry of vector representing World to reference inertial coordinate frame of pose measurement
}
```

bloesch, 2016-06-17:
Just as a first input: your virtual horizon is upside-down, try inverting the camera-imu extrinsics quaternion (changing the sign of x/y/z).

wf34, 2016-06-17:
Thank you. It was excellent cue about the imu position illustration in the top-left. World now is not upside-down, but gyroscope level still does not look right - it's stale now. Keypoints live a bit longer now, too. Yet it still fails within 15 secs.

I am attaching the data I use and the conversion script. But I strongly sense the mixup in quaternion conventions.
That's the data from KITTI:
[calib_imu_to_velo.txt](https://github.com/ethz-asl/rovio/files/321055/calib_imu_to_velo.txt)
[calib_velo_to_cam.txt](https://github.com/ethz-asl/rovio/files/321056/calib_velo_to_cam.txt)
[readme.txt](https://github.com/ethz-asl/rovio/files/321058/readme.txt)

and the converter script:
```
from __future__ import print_function

import numpy as np
import spiceypy

def print_calib(q, p):
  print('''
            qCM_x {0:0.10f};
            qCM_y {1:0.10f};
            qCM_z {2:0.10f};
            qCM_w {3:0.10f};
            MrMC_x {4:0.10f};
            MrMC_y {5:0.10f};
            MrMC_z {6:0.10f};
        '''.format(q[1], q[2], q[3], q[0],
                   p[0], p[1], p[2]))

imu_to_velo_mat =  [[9.999976e-01, 7.553071e-04, -2.035826e-03],
                    [-7.854027e-04, 9.998898e-01, -1.482298e-02],
                    [2.024406e-03, 1.482454e-02, 9.998881e-01]]
imu_to_velo_mat = np.array(imu_to_velo_mat)

velo_to_cam = [[7.533745e-03, -9.999714e-01, -6.166020e-04],
               [1.480249e-02, 7.280733e-04, -9.998902e-01],
               [9.998621e-01, 7.523790e-03, 1.480755e-02]]
velo_to_cam = np.array(velo_to_cam)

t_VI = np.array([-8.086759e-01, 3.195559e-01, -7.997231e-01])
t_CV = np.array([-4.069766e-03, -7.631618e-02, -2.717806e-01])

M_VI = np.eye(4, 4)
M_VI[:3, :3] = imu_to_velo_mat
M_VI[:3, 3] = t_VI

M_CV = np.eye(4, 4)
M_CV[:3, :3] = velo_to_cam
M_CV[:3, 3] = t_CV

print('M_CV:\n', M_CV)
print('M_VI:\n', M_VI)

M = np.dot(M_CV, M_VI)

q_QSPICE = spiceypy.m2q(M[:3, :3])
q_eng = q_QSPICE; q_eng[1:] *= -1

imu_as_seen_from_cam = M[:3, 3][:, np.newaxis]
cam_as_seen_from_imu = -np.dot(M[:3, :3], imu_as_seen_from_cam)

print_calib(q_eng, cam_as_seen_from_imu.ravel())
```
Resulting pose:
```
            qCM_x -0.5024298217;
            qCM_y 0.4911566892;
            qCM_z -0.5045851232;
            qCM_w 0.5017202750;
            MrMC_x 0.7296881771;
            MrMC_y -1.0753573166;
            MrMC_z 0.3369260839;

```

longbowlee, 2016-06-20:
Have you guys ever made it run? Looks the translation does not make any sense to me. Is it really that long physically?

wf34, 2016-06-20:
@longbowlee no, but I am going to investigate that a bit more today.
It maybe a little off in some dimensions, but must be close to the truth, as the IMU is placed in the back of the car, while the camera rig is in the middle and ontop.
http://www.cvlibs.net/datasets/kitti/setup.php

burrimi, 2016-06-20:
The issue with these datasets might be, that according to the specification the cameras are triggered by the laser and not the IMU. For rovio precise synchronization of the IMU and cameras is essential, since it is a tightly coupled visual-inertial framework and the time delay is not estimated.

wf34, 2016-06-20:
@burrimi Oh, I see. That most likely is the case. I have also struggled and failed to make rovio work with datasets of my own shooting, which also had the camera and IMU not synchronized. 
Shall I close it then?

closed 2016-06-20 by wf34

longbowlee, 2016-06-21:
@wf34 I see, you guys are playing autonomous car while I am playing drone.