ethz-asl/rovio #66: Add FOV distortion model support for FishEye Lens

ginowu, 2016-06-22:
Hi bloesch
        I am using one FishEye Lens camera, then I had calibrated the extrinsic between IMU and Camera, and also the intrinsic of Camera using the FOV model. So I would like to let rovio also support FOV.

Now I had added supporting functions for FOV model to the "Camera::distort" function. But from the visualied GUI frame, I see the same feature point keeps increasing the number on it, so I think the feature tracking is in bad state, right?

As I understand, Camera::bearingToPixel is doing projection(3D -> 2D), and Camera::pixelToBearing is doing back projection(2D -> 3D), how could these two functions utilize the same "Camera::distort" function? I don't quite understand the logic in Camera::pixelToBearing, especially the variable "du" in the for loop, could you help giving some guidance on this item?

Thanks
Gino

bloesch, 2016-06-22:
bearingToPixel does the projection, distortion, and mapping to pixel coordinates. The back projection does the inverse (to a unit length bearing vector). In this, the undistortion is computed by performing an optimization. This basically searches for the best undistorted coordinates such that if you apply the distortion function you obtain your distorted coordinates. This is solved iteratively and du is the iterative correction on the undistorted coordinates ybar.

Ideally, you would only have to change the distort function, since th undistortion automatically adapts to it.

If the numbers keep increasing this means that something went wrong with the prediction (which heavily relies on distortion/undistortion).

ginowu, 2016-06-23:
@bloesch, thanks for your quick reply. In fact I don't have too much belief of the rotation in the calibrated extrincs, do both this wrong R and not accurate distort function contribute to this issue?

The distort function is copied from Kalibr, should be fine in theory, do you have any advice to debug this issue?

bloesch, 2016-06-27:
It would be easiest if you start with a working version of rovio and then adapt the dirstort function. A wrong R can very quickly lead to a wrong prediction of the feature location and you will not be able to debug the distortion process.

You could maybe check the feature_tracker_node, which is a simple feature tracker (no IMU used). But if I remember right this will not directly use the distortion/undistortion - so you would for instance have to change the prediction step in order to change to bearing vectors, do the prediction there, and then change back to pixel coordinates (line 136 in featureTracker.hpp).

csdncannon, 2016-06-28:
I had manually calculated the JPL quaternion of R, it confirms to the result of Kalibr's result. I had merged the distort function from Kalibr, and also replaced undistort function of Rovio with Kalibr's(just the reverse of distort function). 

Is the optimization method to undistort will bring in performance issue? Seems it is called very frequently.

Now I have my FishEye camera working very well with the old version of Rovio in late January.

@bloesch, thanks for your great help!

bloesch, 2016-07-01:
@csdncannon, the implementation of distort/undistort is currently a bit suboptimal and should be replaced by a look up table.

mhkabir, 2016-11-21:
@csdncannon Would you mind sharing the diff you had to make in order to get the new model working? Thanks!

zhangaigh, 2016-11-30:
@csdncannon I've been working with fisheye camera. However, I can't get better performance. How do you do for fisheye camera? Thanks a lot

liucsg, 2017-11-29:
Hi bloesch!
As I understand, Camera::bearingToPixel is doing projection(3D -> 2D), I do not know what is mean of n.getM(). Could you help giving some guidance on this item?
Thanks