ethz-asl/rovio #67: GPU (CUDA)  acceleration for Rovio

mhkabir, 2016-06-26:
@bloesch I was looking into speeding up Rovio on Nvidia GPUs using CUDA.

Is there a timing profile for Rovio , or would you be able to advise on which sections of Rovio it would be most beneficial to parallelize on the GPU to free up CPU (especially in the stereo case).

bloesch, 2016-06-27:
The timing is easily profiled using the google gperftools.

If I remember right the highest computational load comes from the matrix multiplication in the Kalman filter and I am not too sure how easy it is to parallelize those. Since every feature update is performed separately maybe some operation could be improved there: Some pre-computations or the handling of multiple hypothesis could probably be parallelized.

All in all rovio is still a very leightweight visual odometry framework and I believe that with a proper implementation it could be speed up significantly. Most current computational issues are due to sub-optimal implementation/programming and I will try to improve those in a first step.

Still, delegating some operations to the GPU is an interesting thought and could lead to additional features.

mhkabir, 2016-06-27:
> If I remember right the highest computational load comes from the matrix multiplication in the Kalman filter and I am not too sure how easy it is to parallelize those.

Should be easy to optimize on a GPU. e.g http://www.es.ele.tue.nl/~mwijtvliet/5KK73/?page=mmcuda

>  Most current computational issues are due to sub-optimal implementation/programming and I will try to improve those in a first step.

That would be awesome, yes.

We could also probably parallelize feature tracking and track more features without loading the CPU too much, which would allow more robustness. The stereo left-right matching could also be done very effectively on GPU.

I've almost finished my design of a high-speed IMU-synced stereo system (150fps synced VGA res streams) and it would be interesting to see how fast we could get Rovio going in order to use all that data.
What is the current highest framerate you've tested on?

bloesch, 2016-07-01:
The matrix multiplication are mainly done wihtin the lightweight_filtering library. Both, prediction and update, can be costly. The implementations can be found in Prediction.hpp and Update.hpp. I've heard about the GPU based matrix multiplication, but the additional GPU dependency used to be bit cumbersome for robotics. I guess this might change soon.

I am planning to work on rovio end of July. I will collect ideas for improvements/features beforehand.

Paralelizing the feature tracking is not as trivial since it depends and alters the filter state. Maybe there is some literature on that - I haven't checked it so far.

I am still not too convinced of the stereo setup. One issue originiates from the online calibration, which, in the current filter setup, either relies on a good initial guess or gets computationally expensive (which is the case at the moment). A simple fix here would be to handle the cameras separetely until the calibration has converged to a reliable estimate and to enable cross-measurements only after that.

I haven't test rovio with framerates above 30Hz. While you get increased computational load by the more frequent updates, the uncertainty of the prediction could be decreased which may decrease the costs of certain parts again (smaller search ellipsis).

mhkabir, 2016-08-29:
@bloesch Can you suggest how I can get Rovio to run the prediction at IMU rate, such that I can feed the output directly to vehicle controllers? Around 100Hz would be required for smooth control performance. How did you guys do it?

mhkabir, 2016-08-29:
Also - Are you planning to take up the efficiency improvements within Rovio anytime soon? I wanted to optimize the image alignment to operate in parallel, however wouldn't want to change that in case you plan to refactor large parts of Rovio :)

bloesch, 2016-08-29:
For the first question: I would not recommend running the prediction at a higher rate as this can be rather costly due to the propagation of the imu. Rather you should use the current best estimate and do a simple prediction (without covariance) based on the IMU readings.

Sorry about the delay concerning the re-factoring. I am currently working on a new version of the LWF which should be finished by the end of this week. Afterwards I will start integrating the changes in Rovio and start improving the efficiency. However, I am not sure how fast I will be able to move forward with this and can only hardly guess when this will be finished.

But, if you mainly plan on porting the image alignment to gpu I think that this could be done in parallel. I am not planning to change the alignment itself and I think the interface should be more or less clear. The GPU code basically needs to load the image, the warping matrix, the patch and its coordinates, and returns the reduced! system of equation Ax = b (A is 2x2, x is a delta on the beraing vector, b is the residual vector). In essence getLinearAlignEquationsReduced could be handled on the GPU. Important: the final alignment needs to be done withing the EKF.

mhkabir, 2016-08-29:
Thanks for the help. 

> Rather you should use the current best estimate and do a simple prediction (without covariance) based on the IMU readings.

How can I do this? Can you give me a code example?

bloesch, 2016-08-30:
You can have a look at the evalPrediction() in IMUPrediction.hpp. This can be strongly simplified since you can ignore the noise terms (= 0) and only require the prediction of attitude (qWM), position (WrWM), and velocity (MvM). You can assume that the biases remain constant and you do not need to propagate the features. So in the end you should have a very lightweight prediction.

Also, do not change the filter state. Just take the most actual state and forward propagate it to the current time (make a copy of attitude/position/velocity).

nicolov, 2016-09-02:
Regarding parallelization of the image update, rovio right now repeatedly builds and applies a 2-dimensional residual, one for each tracked feature. I was wondering what was @bloesch opinion about building a 2*N "batch" residual and applying it instead. That would make it possible to parallelize the preprocess step in N independent threads and reduce the CPU load due to matrix operations.

Would a few more iterations make up for the loss of precision of the "batched" update? What about handling outliers?

bloesch, 2016-09-02:
This is a good comment. I would not worry about a reduced accuracy. On the contrary, you could gain a certain amount of robustness as you do not rely on a single feature only at a time.

The motivation for  splitting the updates was mainly speed. Due to the much smaller matrix inversion N * 2-D updates are faster than a single 2N-D update (this was actually done in a previous version). But if the evaluation of the residual could be parallelized it might be worth a try to get back.

mhkabir, 2016-09-13:
@bloesch Finished up the IMU-based forward propagator, works great! Thanks for your tips :)

I'm now looking into adding some sort of a smart watchdog which will re-init the filter in case it diverges. Do you have any tips for what metrics to monitor to detect a possible divergence?

bloesch, 2016-09-19:
I am not sure what the best metric is. Possibilities include:
- Size of velocity covariance
- Number of tracked landmarks
- Number of newly initialized landmarks
- Number of tracking failures

This is not an easy problem as you would like to avoid to many resets. I guess one would have to investigate standard divergence cases and try to find a good metric.

mhkabir, 2016-09-19:
@bloesch Thanks. I had been investigating number of features, velocity covariance and timing factors. I have basic watchdog working, which resets the filter to a known good pose when the limits are triggered multiple times in a short interval. This works great, but will still require more work.

How can I access these in the code ?
1. Number of tracked landmarks
2. Number of newly initialized landmarks
3. Number of tracking failures

bloesch, 2016-09-19:
Tracking history for every landmark is kept in the FeatureStatistics class. You can access the objects through filterState.fsm_.features_[ID].mpStatistics_. There is not yet a functionality to compute the total failure count. Also, you would have to add a counter for the initializations (in commonPostProcess of imgUpdate).

mhkabir, 2016-09-26:
Thanks. Still working on making the watchdog more robust, but had a few good flights today with GPS fusion. 

@nicolov wanted to bring in velocity fusion from external sources as well, which is something would be useful in my case as well. I think the implementation requires some discussion however over how to not do it in a hack way.

nicolov, 2016-09-27:
Just wondering if @bloesch would rather merge a separate `VelocityUpdate` class for velocity measurements or a patch to add velocity info to the existing `PoseUpdate`. I guess the former would be better, as one could potentially use multiple sources of velocity data (i.e. GPS & barometer)

bloesch, 2016-09-27:
I agree with you. Adding a VelocityUpdate would be the nicer solution.

closed 2023-07-17 by mhkabir