ethz-asl/rovio #68: Performance with 10-20m scene depth

nicolov, 2016-07-01:
Hi Michael,
   testing of rovio while hand-holding a custom-made synchronized IMU-camera sensor has been great so far. However, flying at higher altitudes (10-20m) with common automated waypoint-based UAV patterns has shown some divergence problems as mentioned in the paper:

> Furthermore, we also observed a divergence mode for the
presented approach. It can occur when the velocity estimate
diverge, e.g., due to missing motion or too many outliers. The
problem is then, that the filter attempts to minimize the effect
of the erroneous velocity on the bearing vectors by setting
the distance of the features to infinity. This again lowers
any corrective effect on the diverging velocity resulting in
further divergence

In these situations, the feature tracking seemed to work pretty well, it's usually the depth propagation that becomes bad enough over time. Neither the zero-velocity updates nor some tricks I added (like using the barometer altitude or climb speed) have made much impact.

I was wondering if you had done experiments in similar situations or had switched to a stereo setup as stated in the conclusion of the paper.

bloesch, 2016-07-01:
I have not tested rovio at higher altitudes so far. In the end I guess it will depend on the ration between motion and depth of your features.

One issue I have seen with forward looking cameras is that the features tend to stick to the horizon which is often far away and thus does not provide any significant velocity feedback. Here, a solution is to increase the penaltyDistance parameter in order to improve the distribution of the features throughout the image.

I am not sure how much the zero velocity update could help. So far it is meant for situation where everything is absolutely stationary and I doubt that it will get triggered if you are flying. There are a couple of other parameters which could get tuned, e.g., you could reduce the process noise on the feature distances (PredictionNoise.dep).

The best approach would probably be to detect situation with large scene distance and to adapt the parameters accordingly. This is already done for the initialization, which initiatlizes new features at the median of the currently tracked features. Still, other parameter like the noise on the depth could be adapted as well.

Ultimately, if the entire scene is too far away monocular (and even stereo) VIO approaches will always fail without addtional help. Here I see three options:
- you make use of the PoseUpdate in order to provide GPS data or similar to rovio
- you employ the motion model of your UAV to provide addtionaly information to the filter (something we are currently investigating at our lab)
- you add further assumptions to the filter, e.g. that you mean velocity/acceleration is zero.

Still, I think/hope that 10-20m should be feasible in the current setup. Do you have a corresponding dataset? I could try tuning a couple of parameters.

nicolov, 2016-07-04:
I'm actually using a downward facing camera and feature distribution looks great.

I'd actually tweaked the zero velocity update to use the number of recently added features as a proxy for movement, rather than the apparent motion within the frame. It seems to better detect the kind of mostly-hovering situations that tended to cause divergence.

I actually went a step further than reducing the process noise on the depth and actually tried pausing the prediction of the feature depths when the velocity is too low, as those were quite inaccurate anyway (always tending towards higher altitudes than ground truth).

Thanks for the suggestion: I'm already using the barometer to initialize new features (as the ground is pretty flat anyway), but will try the noise as well.

I've already tried GPS input, both position and velocity corrections. It does help the trajectory a great deal (obviously..) but only marginally affects this depth divergence problem. Maybe due to the robocentric setup, corrections to the world position have trouble "flowing back" to the feature states? I've also tried incorporating climb speed from the baro as a vertical velocity component, as you suggested, but this also has a marginal effect.

nicolov, 2016-07-04:
If you want to try a bag, this is pretty representative of typical UAV agriculture/surveying applications. There's both GPS and barometer data as well. The images are compressed, so you'll need an `image_republish` node (see below).

https://db.tt/j0e3v2S5

Camera calibration:

```
image_width: 752
image_height: 480
camera_name: cam0
camera_matrix:
  rows: 3
  cols: 3
  data: [642.165522213, 0.0, 375.47211766, 0.0, 639.779808626, 234.794119093, 0.0, 0.0, 1.0]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data:  [-0.419426943073, 0.188273898725, -0.000976553072974, 2.76869957082e-05, 0.0]
```

Extrinsics:

```
Camera0
{
    CalibrationFile             ; Camera-Calibration file for intrinsics
    qCM_x  -0.99998294779     ; X-entry of IMU to Camera quaterion (JPL)
    qCM_y  0.00472972745206     ; Y-entry of IMU to Camera quaterion (JPL)
    qCM_z  -0.00274881055966     ; Z-entry of IMU to Camera quaterion (JPL)
    qCM_w  0.0020439786779     ; W-entry of IMU to Camera quaterion (JPL)
    MrMC_x 0.00321097330771    ; X-entry of IMU to Camera vector (expressed in IMU CF) [m]
    MrMC_y 0.0307398030293    ; Y-entry of IMU to Camera vector (expressed in IMU CF) [m]
    MrMC_z -0.020124698444    ; Z-entry of IMU to Camera vector (expressed in IMU CF) [m]
}
```

ROS:

```
<node pkg="rovio" type="rovio_node" name="rovio" output="screen" required="true">

    <param name="filter_config"
           value="$(arg cfg_folder)/rovio.info"/>
    <param name="camera0_config"
           value="$(arg cfg_folder)/rovio_cam.yaml"/>

    <remap from="/imu0" to="/imu"/>
</node>

<node name="image_republish" pkg="image_transport" type="republish"
      args="compressed in:=/camera/image_raw raw out:=/cam0/image_raw">
</node>
```

Thanks, I'm leaving this in here in the hope that it'll be useful to someone else as well.

mhkabir, 2016-07-04:
@nicolov, would you be willing to share your modifications to Rovio to add barometer and GPS velocity data?

nicolov, 2016-07-05:
Without any changes to the code, you can already use the baro with a bit of ROS magic, by synchronizing the gps and baro topics and creating a new topic with `x` and `y` from the GPS, and `z` from the baro.

The velocity update is just a few lines of code in the PoseUpdate class. It's important to account for the attitude, since the velocity in rovio's state is in body coordinates, whereas the GPS gives you velocities w.r.t. to a north aligned frame (ENU).

```
y.vel() = state.qWM().rotate(state.MvM()) - meas_.vel()  + noise.vel();
```

bloesch, 2016-07-05:
I agree with @nicolov - this is a relative clean and simple way to introduce the additional measurements. Theoretically the velocity update is not really required as it, again theoretically, does not provide any additional information to the GPS position measurements. This depends a bit on how the GPS velocities are computed: if a low pass filter is employed this adds further information to the measurements (something like smoothness) and will affect rovio's output (might be desirable).

Concerning the previous post: I think the "zero velocity update" should be replaced by a handling of "missing image motion" which would also catch the case where motion is apparent in the IMU readings but not available in the image data (e.g. due to large scene distance). The questions wether a zero velocity update is appropriate in such a case or wether there are more appropriate means of stabilizing the filter - I think the IMU filter literature might be interesting for this.

Pausing the prediction can be a dangereous approach as this might deteriorate the numerical properties of your filter by repetitive update (the condition number of the covariance matrix will get bad, but not sure a fast this happens).

To what extent is the depth divergence a problem? If the scene is far away the depth can only difficultly be estimated (this is a general problem for mono and you are right that it might get worse with the robocentric setup) and the filter should automatically rely more on the GPS data. Or does it eventually lead to the divergence of the full filter state?

Also, thank you very much for the dataset @nicolov. I will have a look at it during the next rovio iteration, which I will be starting by the end of July.

nicolov, 2016-07-05:
The hope behind using the GPS velocity was having a more direct effect on the velocity in the state, which is used to propagate the features.

On the bag I posted, it's hard to get more than 45s-1m of stable operation. Afterwards, the depth starts to climb and eventually corrupts the feature tracking as well (very wide covariance ellipses in the frame). I'm talking about full-on divergence to infinity, from which I can only recover with a complete reset of the state. Again, no problem when the scene is only 1-2 meters away, even for long periods of time.

The filter seems to cope with pausing the prediction for 10-20 seconds, and that obviously stops the divergence since the depths barely change at all without the `velocity * normal` term.

What are you planning for the next version?

nicolov, 2016-07-05:
As an example, these are plots for the bag I posted earlier. After about 40 seconds, the median of the feature depth (green) starts to climb compared to ground truth by the barometer (blue). At this stage, feature tracking usually still works decently (but the trajectory is compromised given the scale problems, as expected).

[image]

As stated in the paper, this behavior is apparently of a diverging nature: after a few more seconds, the depths start growing without bounds and eventually the tracking fails as well (blue dots in the frame).

[image]

bloesch, 2016-07-06:
Thanks for the plots @nicolov. I am a bit surprised that the pose estimation is affected by the diverging feature distances as the feedback should get very small once the features are far away and you still have the GPS measurements - but I might also be missing something. Maybe it is also due to numerical issues.

In the next code iteration I will try to address various issues/features which may include:
- Computational issues which are mainly due to inneficiencies in the code. For instance the initialization of new features is highly inefficient due to the high number of candidates and the expensive computation of scores.
- Improving robustness which involves various methods that have been discussed in the different issue discussions. For instance the detection of missung velocity feedback (due to large scene distance or missing motion) and handling by exploiting some additional assumption (might lead towards zero velocity updates).
- Implement a fault detection and an "intelligent" resetting.
- Time alignment of IMU and vision which seems to be a big source of confusion.

I am planning to open an issue before starting the new iteration in order to collect inputs/comments/wishes from the community.

nicolov, 2016-07-08:
Looking at the robocentric SLAM literature, it seems that a trick to improve performance is to delay prediction of the feature states until *after* the EKF update. This is supposed to be better because the inaccurate IMU measurements are not inverted right away to propagate, adding linearization errors. Instead, the process goes like this:
- Use the motion model to predict the world position and attitude *only*. Concatenate the new states to the existing state vector.
- Perform the ekf update with the image information
- Use the corrected incremental motion to propagate the features states

I was wondering if you thought this would be somehow useful.

I'm sure you're familiar with some of these papers:
- http://www.ki.inf.tu-dresden.de/~wiki/Robotics/Tutorials/SLAM-Summer-2006/Lectures/Tardos/Papers/Caste_JRAS_2006.pdf
- http://www.robots.ox.ac.uk/~lav/Papers/williams_reid_icra2010/williams_reid_icra2010.pdf

bloesch, 2016-07-08:
Thank you for the hint @nicolov , I was not fully aware of the delayed prediction approach. However, in comparison to vision only, the inertial data provide a good estimate of the incremental motion and thus the co-estimation of the incremental motion makes only marginally sense.  Also, the current iterative Kalman filter performs multiple relinearization per landmark and should improve the overall consistency.

Still, it is definitely worth considering alternative filtering approach. For instance the iterations in the IEKF are rather expensive and a more efficient approach would be nice.

nicolov, 2016-07-12:
Thanks for the remark, it made me realize that it was just a trick to save one Kalman iteration. Now that I've finally got the iterative variant to work, the divergence problems are much less severe.

For some reason, the intensity covariance should be much lower than the non-iterative case. I hope that helps solve some of the issues people have been having with the newest revisions.

bloesch, 2016-07-28:
What do you mean with non-iterative case? Did you set maxNumIteration to 1? I doubt that this would actually work, maybe for very slow motions...

nicolov, 2016-07-28:
I meant using `modeEKF` instead of `modeIEKF`. By the way, is there any reason that the indirect error doesn't work very well with the IEKF?

bloesch, 2016-07-28:
Alright, I understand what you did.

It has been a while since I last tested the indirect error. Typically it used to work relative well.  Does it perform well when using the EKF?

nicolov, 2016-08-03:
Yep, I think it may actually be more robust in situations like the bag I posted above. I just can't seem to get it to work well with the IEKF.

In any case, the project I have been using rovio for ended, so I unfortunately won't have a lot of data to test any improvements in the future. Also, congratulations for the new [deep learning paper](https://arxiv.org/abs/1607.07405), it look really cool.

mhkabir, 2016-08-12:
@bloesch Are your plans to work on Rovio still on? I would like to benefit from parallelization wherever possible (e.g image alignment) but I wouldn't like to touch what you're refactoring because it would lead to painful merge issues later.

I'm currently aiming to get 30Hz updates on a quadcore ARM processor, which should definitely be possible with some optimizations. Without any optimizations and in stereo mode, I get a measly 5Hz. I would also like to profile the code, but I haven't used gperftools with ROS before, so would you mind giving some instructions on how you do the profiling?

@nicolov Any advice would be welcome as well! :)

Thanks!

nicolov, 2016-08-12:
There's some info about performance in one of the other issues (number of
features, patch size, etc..)

For the little profiling I'd done, the most time was spent when detecting
new features.  To relieve some of this load, you should increase the
detector threshold in th config file. You'll get 20-40% better frame
rates if I remember correctly.

If you think about it, going from 5 to 20 hz on 4 cores requires perfect
parallelization, which is really hard to achieve in practice.

A common approach in these kinds of systems is to split tracking and new
feature detection in two threads. But that would give you nowhere close to
4x the performance..

mhkabir, 2016-08-12:
@nicolov Thanks for the comments. 

Perfect parallelization isn't practical, but I'm hoping that optimizing the particularly heavy operations for the GPU will yield better raw performance. 

For example, I'm sure that the feature detection can be speeded up quite a bit. I will time the detection performance and go from there. 

Did you use any specific profiling tools in your experiments?

bloesch, 2016-08-12:
Concerning the profiling you could directly start the node executable with the profiling prefixes, or otherwise I think you can put it into the roslaunch file (you will have to look that up). Here you can find some info: https://github.com/ethz-asl/programming_guidelines/wiki/Profiling-Code.

I have started improving the LWF, mainly cleaning up and making it more readable. But this might also improve performance slightly. Currently I am working in a new repo (https://github.com/ethz-asl/generalized_information_filter).

I think some significant performance improvements should be achievable by optimizing some parts of the code. As @nicolov mentioned the detection of new feature is terribly slow and should definitely be improved. Also, maybe sparse matrices could be used in certain parts (not sure about this).

mhkabir, 2016-08-12:
@bloesch Thanks a lot for the resources. The profiling page was super-useful!

Great to hear that you're working on LWF! Will await your results. 

Currently I'm moving over some processing in Rovio to work on the GPU, and feature detection speeds should be improved quite a bit.

mhkabir, 2016-08-14:
@nicolov @bloesch Have it running at > 25Hz with shared-memory GPU acceleration and patchSize = 6.

Changliu52, 2016-08-14:
Hi @mhkabir , that sounds like a major improvement. Very impressive! Which ARM computer you are using? And which branch do you refer to?

mhkabir, 2016-08-15:
@Changliu52 I'm using an Nvidia Tegra X1. I'm using the branch from January + my GPU optimizations.

Changliu52, 2016-08-15:
@mhkabir That's really cool! I am preparing to use Jatson TX1 to run rovio. Do you plan to publish your GPU accelerated version on a separate Fork?