ethz-asl/rovio #72: Error opening file?

sc38514, 2016-07-11:
I'm using Ubuntu 14.04 with ROS indigo.

The launch file is set to run only one camera and the IMU and camera topics are remapped to the corresponding topics in my bag file.

The error shows:

Recording
Storing output to: /path/to/file.bag
terminate called after throwing an instance of 'rosbag::BagIOException' 
 what(): Error opening file: /path/to/file.bag

Anyway to resolve this?

Thanks.

mhkabir, 2016-07-12:
You're supposed to use the actual path to the bagfile instead of "/path/to/file.bag"

sc38514, 2016-07-12:
Actually I just censored the path/file name due to an NDA, but the path to bagfile was specified.
However I don't think reading/playing from my bagfile is causing the error because the error is from opening the file to record:

what(): Error opening file: /some/path/rovio/2016-07-12-01-45-12_25_4_6_1_0.bag

where the folder /some/path/ is where my bags are stored, the /rovio folder I assume is created by rovio node, and 2016-07-12-01-45-12_25_4_6_1_0.bag looks like a timestamped file also generated from the rovio node.

bloesch, 2016-07-12:
I think you have to pre-create the rovio folder manually. this is an issue that should get slved in the future.

andrejpan, 2016-08-01:
When I ran rovio_rosbag_node.launch I also had the same problem. After making rovio folder package started working.

Denfender, 2016-09-29:
I have the same problem with you when running rovio_rosbag_node.launch,how do you solve it ?Thanks

Denfender, 2016-09-29:
@andrejpan hope your advice

bloesch, 2016-09-29:
if you get:
  what(): Error opening file: /some/path/rovio/my.bag
do:
  mkdir /some/path/rovio

Denfender, 2016-09-30:
Thanks for reply! but i doesnot understand. I have put my bag into the path where the launch file inculded.
I think this cannot solve my problem.Does  this bag should put into "rovio" file?@bloesch

bloesch, 2016-10-03:
You might have a different error. Did you try building in debug mode? What error do you get?

Denfender, 2016-10-15:
thank you .it works.I should build in debug mode. thank you !also ,opengl shuold be used for view.thank you @bloesch

bloesch, 2016-10-15:
just be sure to switch back to release mode

closed 2016-10-15 by bloesch

karanchawla, 2016-10-17:
I created the "rovio" folder in my /karan/Downloads/ directory. I still get this error. What am I missing? 

https://snag.gy/60zYoe.jpg

Thank you!

Denfender, 2016-10-22:
@karanchawla ，this is a common problem.you can change the permissions. if it dosenot work ,you should use a absolute path .in a word ,you can also rosplay this bag,not using the  rovio_rosbag_node.launch.

Kyle-ak, 2016-12-12:
@Denfender Could you tell me how you made rovio work? I'm new to ROS, where should I create the "rovio" folder? And I really couldn't get a clue the error I was informed of:

Recording
open file:</home/ak/workspace/rovio_catkin_ws/src/rovio/dataset.bag>
terminate called after throwing an instance of 'rosbag::BagIOException'
  what():  Error reading version line
Aborted (core dumped)

bloesch, 2016-12-15:
@Kyle-ak can you check your bag file? try rosbag info dataset.bag ...

Otherwise, please build in debug mode to get a more details.

Kyle-ak, 2016-12-15:
@bloesch Thanks for your reply, it's indeed the rosbag file... One of the two files is corrupted. Now I can get it running on the _MHMH_03_medium.bag_. However, another problem popped up. I got the warning:

==== Model jacInput Test failed: 74.0615 is larger than 1 at row 1(def_0) and col 0() ====

And the feature tracking part seems fine, but the Scene is static all the time. 

[image]

better-fcl, 2020-04-22:
> I think you have to pre-create the rovio folder manually. this is an issue that should get slved in the future.

thank you!