ethz-asl/rovio #73: error: GLSL ES 3.00 is not supported

blankered, 2016-07-27:
I have a problem when I ran the rovio_node.launch. It shows,
Error compiling shader type 35633: '0:1(10): error: GLSL ES 3.00 is not supported. Supported versions are: 1.10, 1.20, and 1.00 ES
[rovio-2] process has died [pid 4556, exit code 1, cmd /home/uav/vision_ws/devel/lib/rovio/rovio_node __name:=rovio __log:=/home/uav/.ros/log/9c1c037e-5411-11e6-bb65-002170f29c86/rovio-2.log].
log file: /home/uav/.ros/log/9c1c037e-5411-11e6-bb65-002170f29c86/rovio-2*.log
^C[rosout-1] killing on exit
Does anyone know how to solve this problem?

mhkabir, 2016-07-27:
Build it without the OpenGL Scene.

blankered, 2016-07-27:
Thank you. That solved the issue.

blankered, 2016-07-28:
Does this mean I can't use the visualizer?

closed 2016-07-28 by blankered

bloesch, 2016-07-29:
You need support for glsl es 3 which you apparently do not have. This can be limited by your graphics card. Or you "just" have to install it.

A similar vizualization can be achieved with rviz. Most qunatities (features, coordinate frames) are being published by default.