ethz-asl/rovio #74: Compilation Error with Opencv3, no member named 'mpCamera'

blankered, 2016-07-28:
I had to switch my project over to a new laptop which supports opengl ES 3.0. In doing so, I had to start from scratch in gathering the dependencies and supporting packages. The other difference between my successful compilation on my old laptop and new laptop is that I decided to use OpenCV3 on my new laptop with all of the extra modules installed.. I had to make modifications to point the Eigen dependency to Eigen3 but that is about it as far as changes. The log is attached.

[rovio.build.log.txt](https://github.com/ethz-asl/rovio/files/389384/rovio.build.log.txt)

closed 2016-07-28 by blankered

bloesch, 2016-07-29:
This looks like a typo in your Camera.hpp header. Is this why you closed the issue?

blankered, 2016-07-29:
Yes it is.

blankered, 2016-07-29:
After fixing that issue, I run into this issue now.
[rovio.build.log.txt](https://github.com/ethz-asl/rovio/files/390604/rovio.build.log.txt)