ethz-asl/rovio #75: Trouble linking compilation to OpenCV3

blankered, 2016-07-29:
I have been trying to compile ROVIO with OPENCV3. I have successfully compiled ROVIO on an older system with OPENCV2. 

I believe the issue to be with ROVIO being unable to find the correct paths to OPENCV3. It is worth noting that OPENCV2 (libopencv-dev) is installed as a result of mavros-extras package. I tried installing ros-indigo-opencv3 to see if it would fix the pointer issues. I am unsure how to proceed at this point. I have attached my latest build log.

[build.make.008.log.txt](https://github.com/ethz-asl/rovio/files/390769/build.make.008.log.txt)

blankered, 2016-07-29:
I was unable to use opencv3. I removed opencv3 and ros-opencv3 to make it finally compile.

closed 2016-07-29 by blankered

bloesch, 2016-07-29:
I did not try the opencv3 build and I am aware that there might be issues, especially if opencv 2 and 3 are installed concurrently. Issue #32 and #52 are related to this. Some people managed to make it work eventually.