ethz-asl/rovio #76: How do I properly forward imu data to rovio?

blankered, 2016-08-01:
Hi,

I am getting stuck in the configuration stage of rovio trying to forward imu data from my PX4 via mavros to Rovio. I have tried many different configurations in my launch file but I am having no success. I get no indicator that imu data is being forwarded in RQT_Graph. I have Mavros running in another terminal publishing raw imu data to /mavros/imu/data_raw.

Here is my terminal output
`uav_master@MobileGCS:~/vision_ws/src/rovio/launch$ roslaunch rovio_node.launch
... logging to /home/uav_master/.ros/log/b123a4de-5803-11e6-af86-204747c2636d/roslaunch-MobileGCS-17397.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://MobileGCS:43650/

SUMMARY
========

PARAMETERS
 * /cam0/camera_frame_id: usb_cam
 * /cam0/image_height: 480
 * /cam0/image_width: 640
 * /cam0/io_method: mmap
 * /cam0/pixel_format: yuyv
 * /cam0/video_device: /dev/video1
 * /image_view/autosize: True
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /rovio/camera0_config: /home/uav_master/...
 * /rovio/filter_config: /home/uav_master/...
This error is only thrown when I use the following:
    <remap from="/imu0" to="/mavros/imu/data"/>
NODES
  /
    cam0 (usb_cam/usb_cam_node)
    image_view (image_view/image_view)
    rovio (rovio/rovio_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[cam0-1]: started with pid [17415]
process[image_view-2]: started with pid [17416]
process[rovio-3]: started with pid [17417]
[ INFO] [1470070253.664638694]: Using transport "raw"
Set Camera Matrix to:
  958.1       0 274.394
      0 967.598 295.947
      0       0       1
Set distortion parameters (Radtan) to: k1(0.103935), k2(-0.25267), k3(0), p1(0.021415), p2(-0.014961)
Testing Prediction
==== Test successful (5.91687e-08) ====
==== Test successful (5.54897e-08) ====
Testing cameraOutputCF
==== Test successful (6.36613e-08) ====
Testing imuOutputCF
==== Test successful (1.54622e-08) ====
Testing attitudeToYprCF
==== Test successful (4.66041e-08) ====
Testing transformFeatureOutputCT
==== Test successful (1.02961e-08) ====
Testing LandmarkOutputImuCT
==== Test successful (2.26763e-08) ====
Testing FeatureOutputReadableCT
==== Test successful (1.10557e-08) ====
Testing pixelOutputCT (can sometimes exhibit large absolut errors due to the float precision)
==== Model jacInput Test failed: 591.941 is larger than 1 at row 1(def_0) and col 0() ====
  649694  649102
Testing zero velocity update
==== Test successful (3.04311e-10) ====
==== Test successful (1.39778e-10) ====
[ INFO] [1470070253.890565167]: using default calibration URL
[ INFO] [1470070253.890691757]: camera calibration URL: file:///home/uav_master/.ros/camera_info/head_camera.yaml
[ INFO] [1470070253.891664723]: Starting 'head_camera' (/dev/video1) at 640x480 via mmap (yuyv) at 30 FPS
-- Filter: Initializing using accel. measurement ...
-- Filter: Initialized at t = 1470070253.92
[ WARN] [1470070254.176357929]: unknown control 'white_balance_temperature_auto'

[ WARN] [1470070254.179373808]: unknown control 'focus_auto'

OpenCV Error: Assertion failed (scn == 1 && (dcn == 3 || dcn == 4)) in cvtColor, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/color.cpp, line 3789
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/color.cpp:3789: error: (-215) scn == 1 && (dcn == 3 || dcn == 4) in function cvtColor

[rovio-3] process has died [pid 17417, exit code -6, cmd /home/uav_master/vision_ws/devel/lib/rovio/rovio_node /imu0:=/mavros/imu/data __name:=rovio __log:=/home/uav_master/.ros/log/b123a4de-5803-11e6-af86-204747c2636d/rovio-3.log].
log file: /home/uav_master/.ros/log/b123a4de-5803-11e6-af86-204747c2636d/rovio-3*.log
`

This error is only thrown when I use the following:
    <remap from="/imu0" to="/mavros/imu/data"/>

If I use:
  <param name="imu0_topic_name" value="/imu/data_raw"/>
  <param name="cam0_topic_name" value="/cam0/image_raw"/>
  <remap from="/imu/data_raw" to="/mavros/imu/data_raw"/>

I then I get:
`uav_master@MobileGCS:~/vision_ws/src/rovio/launch$ roslaunch rovio_node.launch
... logging to /home/uav_master/.ros/log/b123a4de-5803-11e6-af86-204747c2636d/roslaunch-MobileGCS-17712.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://MobileGCS:49584/

SUMMARY
========

PARAMETERS
 * /cam0/camera_frame_id: usb_cam
 * /cam0/image_height: 480
 * /cam0/image_width: 640
 * /cam0/io_method: mmap
 * /cam0/pixel_format: yuyv
 * /cam0/video_device: /dev/video1
 * /image_view/autosize: True
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /rovio/cam0_topic_name: /cam0/image_raw
 * /rovio/imu0_topic_name: /imu/data_raw

NODES
  /
    cam0 (usb_cam/usb_cam_node)
    image_view (image_view/image_view)
    rovio (rovio/rovio_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[cam0-1]: started with pid [17730]
process[image_view-2]: started with pid [17731]
process[rovio-3]: started with pid [17732]
[ INFO] [1470070687.099686439]: Using transport "raw"
Set Camera Matrix to:
  958.1       0 274.394
      0 967.598 295.947
      0       0       1
Set distortion parameters (Radtan) to: k1(0.103935), k2(-0.25267), k3(0), p1(0.021415), p2(-0.014961)
Testing Prediction
==== Test successful (5.91687e-08) ====
==== Test successful (5.54897e-08) ====
Testing cameraOutputCF
==== Test successful (6.36613e-08) ====
Testing imuOutputCF
==== Test successful (1.54622e-08) ====
Testing attitudeToYprCF
==== Test successful (4.66041e-08) ====
Testing transformFeatureOutputCT
==== Test successful (1.02961e-08) ====
Testing LandmarkOutputImuCT
==== Test successful (2.26763e-08) ====
Testing FeatureOutputReadableCT
==== Test successful (1.10557e-08) ====
Testing pixelOutputCT (can sometimes exhibit large absolut errors due to the float precision)
==== Model jacInput Test failed: 591.941 is larger than 1 at row 1(def_0) and col 0() ====
  649694  649102
Testing zero velocity update
==== Test successful (3.04311e-10) ====
==== Test successful (1.39778e-10) ====
[ INFO] [1470070687.321380506]: using default calibration URL
[ INFO] [1470070687.321540739]: camera calibration URL: file:///home/uav_master/.ros/camera_info/head_camera.yaml
[ INFO] [1470070687.322607110]: Starting 'head_camera' (/dev/video1) at 640x480 via mmap (yuyv) at 30 FPS
[ WARN] [1470070687.607545284]: unknown control 'white_balance_temperature_auto'

[ WARN] [1470070687.613731130]: unknown control 'focus_auto'
`

I was using this [https://github.com/ethz-asl/rovio/issues/53](url)
[https://github.com/ethz-asl/rovio/issues/53](url)source as an indicator that I am still do not have not the imu configuration properly set.

Thanks for any advice.

blankered, 2016-08-01:
Looks as though I have bigger issue really..

`-- Filter: Initializing using accel. measurement ...
-- Filter: Initialized at t = 1470071886.44
OpenCV Error: Assertion failed (scn == 1 && (dcn == 3 || dcn == 4)) in cvtColor, file /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/color.cpp, line 3789
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/buildd/opencv-2.4.8+dfsg1/modules/imgproc/src/color.cpp:3789: error: (-215) scn == 1 && (dcn == 3 || dcn == 4) in function cvtColor`

closed 2016-08-01 by blankered

mhkabir, 2016-08-01:
Your images are monochrome?