ethz-asl/rovio #79: build error (Eigen missing)

amiltonwong, 2016-08-05:
Hi, all,

I got the following build error. It seems Eigen is missing in my system (ROS Indigo). Could someone suggest me how to fix it? Thanks

```
root@milton-ThinkPad-T450:~/catkin_ws# catkin build rovio --cmake-args -DCMAKE_BUILD_TYPE=Release
-------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /root/catkin_ws
-------------------------------------------------------
Source Space:       [exists] /root/catkin_ws/src
Log Space:          [exists] /root/catkin_ws/logs
Build Space:        [exists] /root/catkin_ws/build
Devel Space:        [exists] /root/catkin_ws/devel
Install Space:      [unused] /root/catkin_ws/install
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------
[build] Found '4' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Warning: devel space setup files have an unknown origin.
Starting  >>> rovio                                                            
_______________________________________________________________________________
Errors     << rovio:cmake /root/catkin_ws/logs/rovio/build.cmake.006.log       
CMake Error at /root/catkin_ws/src/rovio/lightweight_filtering/CMakeLists.txt:15 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

cd /root/catkin_ws/build/rovio; catkin build --get-env rovio | catkin env -si  /usr/local/bin/cmake /root/catkin_ws/src/rovio --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/root/catkin_ws/devel/.private/rovio -DCMAKE_INSTALL_PREFIX=/root/catkin_ws/install -DCMAKE_BUILD_TYPE=Release; cd -
...............................................................................
Failed     << rovio:cmake          [ Exited with code 1 ]                      
Failed    <<< rovio                [ 0.0 seconds ]                             
[build] Summary: 0 of 1 packages succeeded.                                    
[build]   Ignored:   3 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 0.2 seconds total.                                            
root@milton-ThinkPad-T450:~/catkin_ws# 
```

csdncannon, 2016-08-05:
sudo apt-get install libeigen3-dev

amiltonwong, 2016-08-05:
@csdncannon ,

Still couldn't pass it. libeigen3-dev is already installed.

bloesch, 2016-08-05:
Usually you should have a FindEigen3.cmake in /usr/share/cmake-2.8/Modules. If you don't have it you can try creating it with the following content:

https://github.com/ethz-asl/kindr/blob/master/cmake/FindEigen.cmake

mhkabir, 2016-08-06:
Actually we should add a pkg-config fallback to lightweight_filtering and Rovio for Eigen detection. 

That's how I fixed this locally, because it couldn't find the FindEigen3.cmake for some reason, even though it was there.

andrejpan, 2016-08-18:
I am not `cmake` expert but this helped me successfully building the `Rovio`. Same update was necessary for `lightweight_filtering` as well.
```
- find_package(Eigen3 REQUIRED) 
+ find_package(PkgConfig)
+ pkg_search_module(Eigen3 REQUIRED eigen3)
````

bloesch, 2016-08-19:
The issue is that depending on the Eigen installation some setups do not have a retrievable FindEigen3.cmake. One of the easiest fixes would be to provide such a file with the rovio distribution. While I am not the opinion that this is the proper solution, I am currently considering adding this file.

closed 2018-01-11 by bloesch

Lxiangyue, 2020-03-20:
I have the same question with you, and this question has been solved now.

-  find_package(Eigen3 REQUIRED)
+ include_directories("/usr/include/eigen3")