ethz-asl/rovio #80: Better stability for fast rotations

jasonwurzel, 2016-08-19:
Hi, I'm having problems with keeping tracking during fast movements. Say my system faces one way and suddenly I rotate by 90 or 180 degrees in any direction so that no current feature is visible anymore and new features have been found. This always results in tracking failure and I have to restart the filter. 
Are there any parameters I can tweak, besides having good hardware?

bloesch, 2016-08-19:
In my experience fast rotations are well handled by rovio. This might however depend on the employed IMU and quickly become a problem if the gyroscope gets saturated.

Saturation is probably the only issue as it might end up with strongly missaligned gravity. If this really happens, I do not see any way around re-initializing the filter (assuming that the visual motion is so fast that feature cannot be tracked).

If the issue is not related to saturation you could try increasing the acc/gyr prediction noise.

jasonwurzel, 2016-08-23:
I did some more experimenting and it is not the fast rotations that are problematic. One problem is, if the lighting changes suddenly (switching from a view of a window to a dark corner in the room). But I guess this can be dealt with with better camera configuration.
The other thing though is that if the system runs for some time and then I just place it on the table and there's some large vibration like banging the fist on the table, the filter quickly diverges and I have to restart. 
I'll try to follow your advice and increase the prediction noises!

bloesch, 2016-08-23:
Auto-exposure usually works fairly well. I also observed the issues with sharp impacts, e.g. if the sensor setup is placed roughly on the table and I am no too sure why this is hapening. Maybe the accelerometer gets saturated. I will have to investigate this issue at some point.

jasonwurzel, 2016-08-30:
Thank you for your support, I'll close this issue...

closed 2016-08-30 by jasonwurzel