ethz-asl/rovio #81: rovio with high precision

jasonwurzel, 2016-08-19:
Maybe someone can help me out: My setting is the following: I need high precision orientation in a relatively small space (maybe 1m³). Are there any special parameters I can tweak to yield higher precision, i.e. what parameters are the most influential for precision?
Thank you!

bloesch, 2016-08-19:
Do you need to support 360 degree of rotation? One issue is that features get very quickly deleted if they leave the field of view. So if you want to achieve long-term accuracy you will have to avoid loosing features (this might be achieved by reducing the pruning factors).

Otherwise, best accuracy typically emerges if the noise parameters best fit the actual system. You could try identifying certain noise parameters. For instance the IMU noise parameters can be identified by using "Allan plots".

jasonwurzel, 2016-08-23:
Hi, 
thank you for your comments! 
No, I don't need 360 degrees of rotation, at least not all at once. What I'm having problems is, if I supply the noise parameters in kalibr, then doing the imu-cam calibration I never get the real imu-cam extrinsics but smaller ones (factor of 10 or more). For example my camera lense is about 0.05 m in y direction from the imu, but as a result I get 0.00005 m as y. Now but if I use those (wrong) parameters (with doVECalibration=false), rovio behaves pretty good. Yet if I approximate the IMU to Camera vector by measuring by hand, the system is VERY unstable.
I took the IMU noise parameters from the allan deviation plots of the supplied datasheet.
Another thing is, if I measure traveled distance on the one hand I have large errors (traveled distance 10 cm, measured 30 cm) but on the other hand it is repeatable. So if I measure multiple times I always get the same value with pretty good accuracy...

Which are the pruning factors?

mhkabir, 2016-08-23:
@jasonwurzel, what IMU are you using?

jasonwurzel, 2016-08-23:
@mhkabir an Epson G364

bloesch, 2016-08-23:
Something doesn't sound right. You should definitely not get such large errors. And if kalibr is also struggling then this is an additional hint that something is not right. Did you doublecheck the units of your IMU measurements? The rotational rate should be measured in rad/s (not degree) and the acceleration should be measured in m/s^2 (not in G). Thus, if your IMU is gravity aligned with the z-axis pointing up you should read (0, 0, +9.81) from your accelerometer.

If you used the Allan plots for tuning the noise parameters (Prediction.att, Prediction.gyb, Prediction.vel, Prediction.acb) then this should usually yield good results. Again, be sure to properly convert the units (they are given as comments in the info file).

Again, bad time alignment can also cause strong performance drops in kalibr/rovio, but I guess you solved the time synchronization on your system.

jasonwurzel, 2016-08-26:
My units are correct, yes.
Just to be sure I'm not making any errors converting the noise parameters: How do I convert those to the desired units (e.g. rad/s to rad^2/s^3)?

bloesch, 2016-08-27:
If you specify a continuous-time white noise you typically have (measurement unit)/sqrt(Hz). For instance for the rotation rate this will be rad/s/sqrt(Hz). This also matches what you receive from the Allan Variance plot. Furthermore, many IMU manufacturers provide their specifications in those units as they are independent of the selected sampling time. So you basically have to square your values. This is also valid for the biases. So if you have the rad/s^2/sqrt(Hz) white noise covariance parameter of you gyroscope bias rate then you get rad^2/s^3 after squaring.

You should really try to get a good kalibr output. While the accuracy of the calibration can strongly depend on the performed calibration motion you should achieve translational errors of less than 1cm. If you observe large errors then something is stange. Also, if you enable the online calibration in rovio and excite with enough rotational motion, what can you observe for the translational extrinsics values? Do they converge to similar values? Are they within a reasonable range?

jasonwurzel, 2016-08-29:
ok, that much is clear, thank you. but I still have a problem with the calculation of acb, which is in m^2/s^5. My "starting" units are in m/s^2/sqrt(Hz) for the accelerometer. Squaring doesn't get me there... Excuse me if those are pretty basic math questions as this is usually not my domain of expertise :-)

bloesch, 2016-08-29:
Typically what you have in m/s^2/sqrt(Hz) is the noise density of the accelerometer which results in the random walk on the velocity (thus Prediction.PredictionNoise.vel with m/s^3). The random walk you have on the accelerometer values (accelerometer bias) is also driven by white noise, but with a power density in m/s^3/sqrt(Hz). Often this specification is not provided by IMU manufacturers and has to be derived from the Allan Plots (+1/2 ramp) or hand-tuned.

I agree that all of this is not entirely clear. I will add an entry in the wiki when I find some time.

jasonwurzel, 2016-08-30:
Ok, I updated the parameter according to your comments and now it's starting to look really good. There's still room for optimizations but only minor tweaking of the parameters are needed. As a conclusion, in the end the things that were crucial for good performance: setting the noise parameters according to your instructions, having the right imu-cam extrinsics, and most important TIMING between the imu and the cam messages. I think I can close this issue. Thank you very much for the support!

closed 2016-08-30 by jasonwurzel