ethz-asl/rovio #82: Rovio always fails to compile the first time on a new computer

ZacharyTaylor, 2016-08-22:
This is something I have seen on a few systems now and am unsure what is causing it. If you create a new catkin_ws and build rovio it will always fail the first time with the error:

Errors     << rovio:make /home/zac/catkin_ws/logs/rovio/build.make.000.log     
In file included from /home/zac/catkin_ws/src/rovio/src/rovio_node.cpp:40:0:
/home/zac/catkin_ws/src/rovio/include/rovio/RovioNode.hpp:50:34: fatal error: rovio/SrvResetToPose.h: No such file or directory
 #include <rovio/SrvResetToPose.h>
                                  ^
compilation terminated.
make[2]: *** [CMakeFiles/rovio.dir/src/rovio_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/rovio.dir/all] Error 2
make: *** [all] Error 2
cd /home/zac/catkin_ws/build/rovio; catkin build --get-env rovio | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

However, if the build is immediately rerun everything builds fine and the error never appears again.

bloesch, 2016-08-23:
I will have a look at this at some point. If anybody else knows the fix for this, a quick note would be welcome.

nicolov, 2016-08-23:
The usual fix is splitting the service definitions in a new (otherwise empty) package that rovio would depend on. The convention would be `rovio_msgs` or `rovio_srvs`. I can push up a PR if @bloesch wants.

andre-nguyen, 2016-08-23:
I don't think you really need to split it off. Basically the problem comes from not making the executable depend on the messages.

However I can't seem to find a good source on the correct way to fix it. It's either this

```cmake
add_dependencies(something_node  ${catkin_EXPORTED_TARGETS})
```

or something with `gencpp``.

nicolov, 2016-08-23:
Nice to know, thanks. It seems that the first option would be the first one ([link](http://docs.ros.org/kinetic/api/catkin/html/howto/format2/cpp_msg_dependencies.html))

bloesch, 2016-08-24:
integrated this in #85. Thanks for the hints!

Femer, 2016-09-13:
Hi,
I get the same error, even though I'm using the latest version of master with the dependency on the exported tag:

Errors     << rovio:make /home/marco/catkin_ws/logs/rovio/build.make.000.log                                                                                                              
In file included from /home/marco/catkin_ws/src/rovio/src/rovio_node.cpp:40:0:
/home/marco/catkin_ws/src/rovio/include/rovio/RovioNode.hpp:50:34: fatal error: rovio/SrvResetToPose.h: No such file or directory
 #include <rovio/SrvResetToPose.h>
                                  ^
compilation terminated.
make[2]: *** [CMakeFiles/rovio.dir/src/rovio_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/rovio.dir/all] Error 2
make: *** [all] Error 2
cd /home/marco/catkin_ws/build/rovio; catkin build --get-env rovio | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

Again, as Zac mentioned, if I compile rovio again after the error everything's fine.

closed 2016-09-26 by bloesch

nico-nutonomy, 2017-01-18:
I am still getting this error. What about splitting messages to a rovio_msgs package? I can submit the patch.

nicolov, 2017-01-18:
I'm still getting this error. What about factoring messages out to a rovio_msgs package? I can submit the patch.

marco-tranzatto, 2017-01-20:
Same problem here, I still need to compile rovio two times in a new catkin workspace

helenol, 2017-01-20:
@marco-tranzatto or anyone else, can you upload exactly the error output when it fails on a clean install? (Or link to a jenkins build that failed due to this)
Having trouble reproducing it locally.

commit 6300d0f6 (2016-09-26): Merge pull request #88 from nicolov/feature/cmakelists

Fix #82
commit 7d528ea0 (2016-09-26): Fix #82