ethz-asl/rovio #83: Segmentation fault (core dumped) / process has died /

BastianJaeger, 2016-08-22:


bloesch, 2016-08-23:
Not too sure about this error. Could you compile in Debug mode? Also, if using the rovio_rosbag_loader please be sure to create the "rovio" folder beforehand (see #72).

nicolov, 2016-08-23:
I had similar issues with 16.04. In my case, the fault was with Eigen vectorization and SSE instructions. I think I ended up disabling static alignment in the `CMakeLists`, as adding a few more `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` wasn't enough.

BastianJaeger, 2016-08-24:
Thank you for the reply.

@bloesch I updated the paths in the .launch file, so I don't see a 'rovio' path needed to create. All paths I used in the .launch file (see post 1) do exist. And since the gdb output says something about `LWF::FilterState` I consider the problme more in the direction of lightweight_filtering or its include. 

@nicolov Eigen it headers only, so what static alignment did you disable? Not sure what to edit in CMakeLists.txt.

nicolov, 2016-08-24:
Try adding this to rovio's CMakeLists. Remember to `catkin clean rovio` beforehand, so that CMake is forced to run again.

```
add_definitions(-DEIGEN_DONT_ALIGN_STATICALLY)
```

The above works on my i7 workstation, but I also had to add the following for Atom CPUs:

```
# add_definitions(-DEIGEN_DONT_VECTORIZE)
```

Do run benchmarks, as I remember these flags decreased performance (as expected).

BastianJaeger, 2016-08-24:
@bloesch After overcoming the main error I understood you comment. Got it & thanks!

@nicolov Thank you very much! The rosbag example is running now! I am still looking for a better solution, because performance is one of the highest demands for me.

I will reply if I got over this eigen "unpleasantness"!

nicolov, 2016-08-24:
Glad to know it worked for you as well. Do measure stuff before worrying though: IIRC the issue only involved pairs of double values that should fit into a single 128 bit register for SSE operations.

The second line is more heavyweight and resulted in a 10-20% slowdown on the Atom CPU that needed it on 16.04.

Hopefully, Eigen 3.3beta2 could solve the issue, but I haven't tried yet.

closed 2017-09-19 by bloesch