ethz-asl/rovio #84: Interesting new stereo cam + IMU offering

nicolov, 2016-08-23:
I've just found this interesting off-the-shelf product with a global shutter stereo camera and integrated IMU. Now that the the VI sensor is no longer for sale, it could be an easy solution for rovio, at $249.

[link](https://www.e-consystems.com/3D-USB-stereo-camera.asp)

While the cameras are synchronized, the spec sheet doesn't say anything about the IMU sync. However, since everything is delivered over the same USB3 connection, you'd think the time delay would be pretty stable, and thus correctable.

mhkabir, 2016-08-24:
Nice! I like it :)

Changliu52, 2016-08-28:
This is perfect!

closed 2016-09-19 by bloesch

gajena, 2018-08-17:
Have anyone tried this one?

mzahana, 2018-10-03:
Any one tried this camera?

jm677zz, 2018-10-04:
I have tried this camera. The IMU and the frame time is not synced, they provide few applications based on OpenCV to evaluate the camera and IMU.

nicolov, 2018-10-04:
Yeah, the IMU is out of sync, and sometimes you get duplicate samples. I had to write my own IMU driver as the sample code was not great.

I also found this other option, but haven’t tried it yet: https://m.seeedstudio.com/productDetail/2873

mzahana, 2018-10-04:
@nicolov thanks for the reply.

For the Optor cam you mentioned. I could not find it's corresponding SDK and its ROS wrapper although it's claimed to have them.

I have not tried that cam, so I don't have feedback about it either.

mzahana, 2018-10-04:
After some search, I found the Optor SDK.
https://github.com/optor-vis/vi-stereo-v1