ethz-asl/rovio #87: Run rovio with stereo camera

zhangaigh, 2016-09-21:
Hello, I would like to make rovio work with stereo camera. Now I use Euroc dataset. I build rovio with the following command:
catkin build rovio --cmake-args -DCMAKE_BUILD_TYPE=Release -DMAKE_SCENE=ON -DROVIO_NCAM=2

After that, rovio can work with multiple camera. However, sometimes rovio could fail which may be caused by filter divergence. I tested rovio with two data set such as MH_05_difficult and MH_01_easy. The first data can work well. However, rovio doesn't work on the second data.  I found that when one camera is utilized, it could work well on both data. 

Could you guys have any ideas for that?
thanks a lot
By the way, rovio is very cool. Thank the authors  for the  impressive contribution.

bloesch, 2016-09-21:
The stereo implementation is a bit tricky. One issue is that if the camera-IMU extrinsics are not very accurate, inconsistencies can quickly accumulate. For this reason the noise parameters were increased in order to allow a potentially large initial extrinsics error. While this works in some cases, this process is very sensitive to (stereo) outliers which can quickly lead the extrinsics to converge to a wrong value.

There are two options to resolve this: either you turn of the cross camera measurements by setting 'useCrossCameraMeasurements' to false. This avoids outliers during the convergence phase and thereby improves stereo performance. On the otherhand a specific landmark will only be tracked in a given camera frame. Ideally this could be adapted such that a landmark automatically turns on the cross camera measurements as soon as its distances has converged.

The other option is to reduce noise on the camera-imu extrinsics ("vep" for linear extrinsics, "vea" for rotational extronsics) or to entirely disable the online calibration (doVECalibration). However, this relies on a very accurate extrinsic calibration of your stereo setup.

zhangaigh, 2016-09-22:
@bloesch  thank you very much! I am gona try your advice.

closed 2018-01-11 by bloesch