ethz-asl/rovio #89: How to run without ros

JimmyTeng, 2016-09-27:
hi,i want to run rovio without ROS, if i refactor, what should I force?
PS : I tried to refactor it but it crash in OutlierDetectionBase check for assertion:
 rovio_node: /usr/include/eigen3/Eigen/src/Core/Block.h:123: Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::Block(XprType&, Eigen::Index) [with XprType = const Eigen::Transpose<const Eigen::Block<const Eigen::Matrix<double, 2, 1>, -1, -1, false> >; int BlockRows = 1; int BlockCols = -1; bool InnerPanel = true; Eigen::Index = long int]: Assertion `(i>=0) && ( ((BlockRows==1) && (BlockCols==XprType::ColsAtCompileTime) && i<xpr.rows()) ||((BlockRows==XprType::RowsAtCompileTime) && (BlockCols==1) && i<xpr.cols()))' failed.

bloesch, 2016-09-27:
Most of rovio is ros-free. I think you can basically extract all filter related stuff from RovioNode.hpp and put it into a standalone cpp.

Not sure what is causing your error. Looks like some unintialized dynamic eigen matrix. Can you backtrack?

pangfumin, 2017-01-28:
@JimmyTeng  I have modified a Ros-free version. Maybe it can help you .
https://github.com/pangfumin/Rovio_NoRos

closed 2017-02-13 by bloesch