ethz-asl/rovio #93: Add a simple demo example

rdube, 2016-10-20:
Would it be possible to add a simple demo example for using rovio eg. with the euroc dataset? It can be very short, just the launch files or nodes which needs to be launched. Is that information already somewhere? Thanks!

burrimi, 2016-10-20:
I think you can adapt this launchfile to point to the correct bag location.
https://github.com/ethz-asl/rovio/blob/master/launch/rovio_rosbag_node.launch

burrimi, 2016-10-20:
it looks like also the default launchfile uses the euroc dataset calibration.
launch this
https://github.com/ethz-asl/rovio/blob/master/launch/rovio_node.launch 
and play the bag in another terminal

rdube, 2016-10-20:
Yes thanks for the info! We will give it another try! Would it help to add this short piece of information in the readme somewhere?

burrimi, 2016-10-21:
it's already in the readme :)

> Euroc Datasets
> 
> The rovio_node.launch file loads parameters such that ROVIO runs properly on the Euroc datasets. The datasets are available under: http://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets

rdube, 2016-10-21:
Fair enough! Thanks! ;)

closed 2016-10-21 by rdube

Kyle-ak, 2016-12-15:
@burrimi Could you tell me how to run rovio on the Euroc dataset? I launched the rovio_node, and then rovio_rosbag_node, but always got the error msg:
      terminate called after throwing an instance of 'rosbag::BagIOException'
      what():  Error reading version line
I have already changed the path in the "rovio_rosbag_node.launch",  still this error keeps coming up. I'm quite new to ros, and I really couldn't find much info about this error.

bloesch, 2016-12-15:
See answers in #72