ethz-asl/rovio #94: Segfault on Ubuntu 16.04

ZacharyTaylor, 2016-10-25:
Currently rovio segfaults on startup on Ubuntu 16.04. I have tried it on both mine and Alex's computer and hit the same issue.

If I compile rovio in debug mode it seems that the issue is with the alignment in Eigen as it then gives the error:

rovio_node: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.

I then ran it through gdb and got the following:

Excess command line arguments ignored. (__name:=rovio)
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/z/catkin_ws/devel/lib/rovio/rovio_node...done.
/home/z/.ros/__ns:=: No such file or directory.
Starting program: /home/z/catkin_ws/devel/lib/rovio/rovio_node 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe54d9700 (LWP 30105)]
[New Thread 0x7fffe4cd8700 (LWP 30106)]
[New Thread 0x7fffdffff700 (LWP 30107)]
[New Thread 0x7fffdf7fe700 (LWP 30112)]
rovio_node: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.

Thread 1 "rovio_node" received signal SIGABRT, Aborted.
0x00007ffff1aea428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff1aea428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff1aec02a in __GI_abort () at abort.c:89
#2  0x00007ffff1ae2bd7 in __assert_fail_base (fmt=<optimized out>, 
    assertion=assertion@entry=0x133a2c8 "(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAss"..., 
    file=file@entry=0x133a198 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", line=line@entry=128, 
    function=function@entry=0x1343b80 <Eigen::internal::plain_array<double, 4, 0, 32>::plain_array()::__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]")
    at assert.c:92
#3  0x00007ffff1ae2c82 in __GI___assert_fail (
    assertion=0x133a2c8 "(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (31)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAss"..., 
    file=0x133a198 "/usr/include/eigen3/Eigen/src/Core/DenseStorage.h", line=128, 
    function=0x1343b80 <Eigen::internal::plain_array<double, 4, 0, 32>::plain_array()::__PRETTY_FUNCTION__> "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 32>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]") at assert.c:101
#4  0x0000000001067bad in Eigen::internal::plain_array<double, 4, 0, 32>::plain_array (this=0x7fffc38752f0) at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:128
#5  0x000000000107f754 in Eigen::DenseStorage<double, 4, 2, 2, 0>::DenseStorage (this=0x7fffc38752f0) at /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:187
#6  0x00000000010684f6 in Eigen::PlainObjectBase<Eigen::Matrix<double, 2, 2, 0, 2, 2> >::PlainObjectBase (this=0x7fffc38752f0) at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:457
#7  0x000000000104ef52 in Eigen::Matrix<double, 2, 2, 0, 2, 2>::Matrix (this=0x7fffc38752f0) at /usr/include/eigen3/Eigen/src/Core/Matrix.h:259
#8  0x0000000001073e2b in rovio::StateAuxiliary<25u, 4, 6, 1>::StateAuxiliary (this=0x7fffc3875250) at /home/z/catkin_ws/src/rovio/include/rovio/FilterStates.hpp:57
#9  0x000000000105f4be in std::_Head_base<10ul, rovio::StateAuxiliary<25u, 4, 6, 1>, false>::_Head_base (this=0x7fffc3875250) at /usr/include/c++/5/tuple:105
#10 0x000000000105f4da in std::_Tuple_impl<10ul, rovio::StateAuxiliary<25u, 4, 6, 1> >::_Tuple_impl (this=0x7fffc3875250) at /usr/include/c++/5/tuple:353
#11 0x000000000105f513 in std::_Tuple_impl<9ul, LWF::ArrayElement<LWF::QuaternionElement, 0u>, rovio::StateAuxiliary<25u, 4, 6, 1> >::_Tuple_impl (this=0x7fffc3875250) at /usr/include/c++/5/tuple:202
#12 0x000000000105f561 in std::_Tuple_impl<8ul, LWF::ArrayElement<LWF::VectorElement<3u>, 0u>, LWF::ArrayElement<LWF::QuaternionElement, 0u>, rovio::StateAuxiliary<25u, 4, 6, 1> >::_Tuple_impl (this=0x7fffc3875250) at /usr/include/c++/5/tuple:202
#13 0x000000000105f5cb in std::_Tuple_impl<7ul, LWF::ArrayElement<rovio::RobocentricFeatureElement, 25u>, LWF::ArrayElement<LWF::VectorElement<3u>, 0u>, LWF::ArrayElement<LWF::QuaternionElement, 0u>, rovio::StateAuxiliary<25u, 4, 6, 1> >::_Tuple_impl (this=0x7fffc3875250)
    at /usr/include/c++/5/tuple:202
#14 0x000000000105f635 in std::_Tuple_impl<6ul, LWF::ArrayElement<LWF::QuaternionElement, 1u>, LWF::ArrayElement<rovio::RobocentricFeatureElement, 25u>, LWF::ArrayElement<LWF::VectorElement<3u>, 0u>, LWF::ArrayElement<LWF::QuaternionElement, 0u>, rovio::StateAuxiliary<25u, 4, 6, 1> >::_Tuple_impl (this=0x7fffc3875250) at /usr/include/c++/5/tuple:202
#15 0x000000000105f683 in std::_Tuple_impl<5ul, LWF::ArrayElement<LWF::VectorElement<3u>, 1u>, LWF::ArrayElement<LWF::QuaternionElement, 1u>, LWF::ArrayElement<rovio::RobocentricFeatureElement, 25u>, LWF::ArrayElement<LWF::VectorElement<3u>, 0u>, LWF::ArrayElement<LWF::QuaternionElement, 0u>, rovio::StateAuxiliary<25u, 4, 6, 1> >::_Tuple_impl (this=0x7fffc3875250) at /usr/include/c++/5/tuple:202
#16 0x000000000105f6ed in std::_Tuple_impl<4ul, LWF::QuaternionElement, LWF::ArrayElement<LWF::VectorElement<3u>, 1u>, LWF::ArrayElement<LWF::QuaternionElement, 1u>, LWF::ArrayElement<rovio::RobocentricFeatureElement, 25u>, LWF::ArrayElement<LWF::VectorElement<3u>, 0u>, LWF::ArrayElement<LWF::QuaternionElement, 0u>, rovio::StateAuxiliary<25u, 4, 6, 1> >::_Tuple_impl (this=0x7fffc3875250) at /usr/include/c++/5/tuple:202
#17 0x000000000105f73b in std::_Tuple_impl<3ul, LWF::VectorElement<3u>, LWF::QuaternionElement, LWF::ArrayElement<LWF::VectorElement<3u>, 1u>, LWF::ArrayElement<LWF::QuaternionElement, 1u>, LWF::ArrayElement<rovio::RobocentricFeatureElement, 25u>, LWF::ArrayElement<LWF::VectorElement<3u>, 0u>, LWF::ArrayElement<LWF::QuaternionElement, 0u>, rovio::StateAuxiliary<25u, 4, 6, 1> >::_Tuple_impl (this=0x7fffc3875250) at /usr/include/c++/5/tuple:202
#18 0x000000000105f789 in std::_Tuple_impl<2ul, LWF::VectorElement<3u>, LWF::VectorElement<3u>, LWF::QuaternionElement, LWF::ArrayElement<LWF::VectorElement<3u>, 1u>, LWF::ArrayElement<LWF::QuaternionElement, 1u>, LWF::ArrayElement<rovio::RobocentricFeatureElement, 25u>, LWF::ArrayElement<LWF::VectorElement<3u>, 0u>, LWF::ArrayElement<LWF::QuaternionElement, 0u>, rovio::StateAuxiliary<25u, 4, 6, 1> >::_Tuple_impl (this=0x7fffc3875250) at /usr/include/c++/5/tuple:202
#19 0x000000000105f7d7 in std::_Tuple_impl<1ul, LWF::VectorElement<3u>, LWF::VectorElement<3u>, LWF::VectorElement<3u>, LWF::QuaternionElement, LWF::ArrayElement<LWF::VectorElement<3u>, 1u>, LWF::ArrayElement<LWF::QuaternionElement, 1u>, LWF::ArrayElement<rovio::RobocentricFeatureElement, 25u> [... cut ...]
#27 0x0000000001013547 in rovio::RovioFilter<rovio::FilterState<25u, 4, 6, 1, 0> >::RovioFilter (this=0x7fffc3875010) at /home/z/catkin_ws/src/rovio/include/rovio/RovioFilter.hpp:72
#28 0x0000000000ffba94 in main (argc=1, argv=0x7fffffffc308) at /home/z/catkin_ws/src/rovio/src/rovio_node.cpp:97

I tried the simple fix of just putting the EIGEN_MAKE_ALIGNED_OPERATOR_NEW macro in the code for StateAuxiliary, but this seemed to have no effect. I am not really sure where to go from here or which of the other possible issues are causing it.

bloesch, 2016-10-25:
may be related to #83 ?

ZacharyTaylor, 2016-10-26:
Seems like the same issue. The proposed fix of disabling vectorization in Eigen would almost certainly fix it, though it seems a little heavy handed. 

I looked into it a bit more and I believe the issue is that the State class in lightweight filtering creates a std::tuple that contains a rovio AuxiliaryState and this Auxilary class creates Eigen matrices which do not end up being correctly aligned. If this turns out to be the issue you could probably correct this by using the Eigen alignment operator when the tuple is created, however I had trouble finding where this actually happens in the code.

bloesch, 2016-10-26:
I would have tried adding EIGEN_MAKE_ALIGNED_OPERATOR_NEW in StateAuxiliary as well.

The tuple is a member of the State class in lightweight_filtering (State.hpp 511), but I am not sure how much this is going to help.

ZacharyTaylor, 2016-11-04:
Issue now resolved by adding EIGEN_MAKE_ALIGNED_OPERATOR_NEW in a few places in lightweight filterinng. See https://bitbucket.org/bloesch/lightweight_filtering/pull-requests/9/stops-crashing-on-ubuntu-16-via-adding/diff for details

closed 2016-11-04 by ZacharyTaylor

bloesch, 2016-11-04:
Zac, do you want to update the submodule in rovio and create a PR? or should I do it?

bloesch, 2016-11-04:
and thank you very much again for locking into this issue and solving it!

nicolov, 2016-12-27:
I actually still get the segfaults on my 16.04, gcc version 5.4.0 20160609. Has anybody else tested this?

ZacharyTaylor, 2016-12-28:
So the fix I put in stopped in segfaulting at startup. It still segfaults when it gets the first message. I think this is due to the tuple manipulation functions in state adding a eigen matrix, but I got busy with something else and never finished looking into it. Currently I am working round tte issue by disabiling eigen alignment.

nicolov, 2017-01-01:
Ok, we both get the same behavior then.