ethz-asl/rovio #95: the rovio/odometry . linear is nan.

Denfender, 2016-10-27:
hello! now I want test rovio using my camera and imu. i have calibrate the cam-imu by kalibr. the result seems great. but when I run the rovio. the rovio/odometry . linear is nan.
[image]

also ,it seem no feature. hope some advise. the parametric is too much.thank a lot

bloesch, 2016-10-27:
Is it nan from the beginning. I guess something is wrong with your settings. Please double check every step you've done and try to think of possible errors.

For instance your image looks rectified? If you load a proper yaml file for the intrinsic calibration you do not need to rectify the image -  this will be taken care of by rovio.

Furthermore try to use simple test cases, what happens if you leave your sensor setup stationary at the beginning? Try also using the virtual horizon in the top left corner for debugging. Ideally this would be aligned with your real horizon (definitely not the case in your image).

Also, please double check the timing? Are the timestamps provided in the IMU msg roughly similar to the one in the image msgs?

And check the IMU measurements your a passing to rovio. Do the look reasonable? Do you approximately read 9.81 along one of the directions of the accelerometer?

Denfender, 2016-10-28:
thanks for reply! it is useful ,i will try.my image is color picture.so i use the image_proc ro change it to grayscale.  i think there have something wrong with my imu ,i will try .thank  very much

jasonwurzel, 2016-10-28:
Hi, maybe this helps: 
Very basic things I had wrong when starting with rovio:

- acceleration measurements have to be in m/s^2
- gyroscope measurements have to be in rad, not in deg/s
- timestamps of imu + video messages have to at least roughly match each other (+/- 4ms). Of course, for more precision they have to match up even more, but above that delta my odometry output was unusable.

Denfender, 2016-10-30:
thank for your advise!maybe i should try to sychornizer the imu and video.now. i have no hardware synchronization.do you have some advise how to software synchronization. i have try to slow down the imu(50HZ) andVIDEO(20hz)to  alleviate this problem. but it doesnot work.hope you have some advise@jasonwurzel

Denfender, 2016-10-30:
also.the imu +video messages have 14ms deviation.do you kown how to set the deviation in the  "rovio" code?thank you !@jasonwurzel

bloesch, 2016-11-01:
slowing down the msgs will not change anything. the only important thing is that the timestamp are expressed with respect to the same clock. the simplest way to achieve this is to generate/overwrite the timestamps when the messages arrive on your main machine, i.e., you just use rostime::now() for every single message. The timestamps do NOT need to be the same! This is not the most accurate method but should work in most cases.

Denfender, 2016-11-01:
thank for your advise! i will try.thank you !@ bloesch

closed 2018-01-11 by bloesch