ethz-asl/rovio #101: unknown error message

Kyle-ak, 2016-12-23:
When I run the comman `roslaunch rovio rovio_rosbag_node.launch` I always get the error message:

==== Model jacInput Test failed: 74.0615 is larger than 1 at row 1(def_0) and col 0() ====
  -72871.9  -72797.9

I did a little bit search among the previous Issues. It seems that it's related to IMU data? BTW, I managed to get visual features:
[image]

rosbag is downloaded from EUROC dataset, and I already checked it, it's not corrupted.

bloesch, 2016-12-23:
there is a comment above the error mentioning that this may be normal due to the limited precision of the finite difference Jacobian test.

Kyle-ak, 2016-12-26:
@bloesch my bad, didn't read the message carefully. Thanks for your reply, and this awesome project 👍

closed 2016-12-26 by Kyle-ak