ethz-asl/rovio #102: About inverse depth and the bearing vector propagating code

xiaopenghuang, 2016-12-25:
Hi all,
I spent a lot time in understand the following formula,
[image]
But i still cannot get it; Any one can tell me how to derivate the formula, any paper or clue.
Thank you.

bloesch, 2017-01-03:
Those are the minimal derivatives of bearing vector and distance parameter. The derivation of the bearing vector is more involved - a journal paper providing more insights is currently under review. The distance parameter derivative is slightly more intuitiv. If "rho" is the distance parameter and "d" the actual distance you can write:
> drho/dt = drho/dd * dd/dt

with:
> drho/dd = 1/(dd/drho)
> dd/dt = - mu^T v

The last derivative basically states that the change in distance is equivalent to the velocity of the camera projected onto the bearing vector of the landmark of interest.

xiaopenghuang, 2017-01-04:
Thanks , can you estimate how long the journal could publish.

bloesch, 2017-01-04:
difficult to estimate. around 1-2 months.

closed 2017-02-13 by bloesch

xiaopenghuang, 2017-02-14:
@bloesch can i get the journal paper now.

bloesch, 2017-02-14:
still under review.

xiaopenghuang, 2017-06-13:
@bloesch can you provide some clue about bearing vector formula. The journal still under review...