ethz-asl/rovio #105: Rovio_node.launch scene still!!

s05vik, 2017-02-01:
I have integrated the package and dataset. Although rovio_rosbag_node.launch displays the feature tracking, but rovio_node.launch seems to shutdown as soon as the former is launched. Also that, the scene provided by rovio_node.launch remains still.
Kindly tell me if I am missing any steps or doing anything wrong?...
NOTE: I am trying to run the framework on euroc datasets <<MH_easy_01.bag>>
I appreciate your help.

bloesch, 2017-02-01:
you only need to do either or:
- launch rovio_rosbag_node.launch and look at the output
- rovio_node.launch and run the bag separetely using "rosbag play ..."

otherwise you have to instances of rovio....

s05vik, 2017-02-02:
Thank you very much @bloesch  . I am working on an uav with stereo camera (ZED/real sense). And I am trying to create _**octree**_ from **_point clouds_** and implement object avoidance (sort of) algorithm.

Firstly, can rovio be run on stereo config?? Although "rovio_node.cpp" has the option for making the number of camera two, will it work? (as in - extract point clouds/depth info using rovio).

Secondly, is rovio dependent on the hardware mentioned in the dataset page? Will it only perform robustly with eg., **_MEMS IMU (ADIS16448)_**? Can any generic IMU (**_MPU6050_** for instance) be used with rovio?

I'll really appreciate for your help. Thanks once again.

bloesch, 2017-02-02:
1. Rovio can be run on stereo. Although this may not necessarily sound intuitiv, the stereo mode is more sensitive to bad IMU-camera calibration. Also, this mode is slightly less "mature" then the monocular case and I would recomend using the monocular setup at the moment. You can simply set the corresponding cmake argument to 2.

2. You should be able to use different hardware and this has been done a couple of times. However, it is really important that you adapt the parameter (extrinsics, intrinsics, noise) accordingly anf that you take care that your msgs are properly time stamped.

s05vik, 2017-02-02:
Thank you once again @bloesch  It was a great help.

closed 2017-02-13 by bloesch