ethz-asl/rovio #108: Feature tracking performance

greymfm, 2017-02-05:
Hello,
I'm running rovio successfully (I will continue to document my setup here: http://grauonline.de/wordpress/?page_id=2014) in an outdoor scene. On the rovio default options, I reduced initial covariances and prediction noise by factor 10 to get it stable for my low cost sensors (monocular camera and MPU6050 IMU).

1. After 60 seconds of stable run, it is not running stable anymore (transition from green to red boxes in the feature tracker, number of patches increases quickly), although this happens for a scene the camera has already seen at the start and that did work stable there - What rovio options could I try out in such a situation?

2. I noticed the overall drift (over time) is significantly higher than in the rovio YouTube video - what options could I try to reduce the overall drift?

3. Is there a drift-correction planned for rovio or any method to 'learn' and recognize a learned position or is there an recommended way to combine another ROS package with rovio to perform this (any examples)?

Thanks,
Alexander

marco-tranzatto, 2017-02-06:
Hi Alexander,

About your second question I could suggest you use an external pose measurement (such as a GPS position converted to a local ENU plane).
Here there is a short explanation about how to do it:

-) Use [geodetic_utils](https://github.com/ethz-asl/geodetic_utils) to convert absolute GPS coordinates into local ENU positions (set ENU origin wherever you think it's the best);
-) Update section "PoseUpdate" of your [rovio.info](https://github.com/ethz-asl/rovio/blob/master/cfg/rovio.info#L175) file used with Rovio:
	-) set `pos_0`, `pos_1`, `pos_2` to `0.1` into section "UpdateNoise" (the value depends on how much you want to trust your position part of the external pose);
	-) set `MahalanobisTh0` to `120.6511204` (increased the default value by a factor of 10);
	-) set `enablePosition` to `true` (we do want to use the position part of the external pose, i.e. the gps position);
	-) set `enableAttitude` to `false` (we do NOT want to use the orientation part of the external pose);
	-) set `noFeedbackToRovio` to `false` (we do want Rovio to benefit from the external pose);
	-) set `doInertialAlignmentAtStart` to `true` (InertialAlignment will take care of aligning Rovio world correctly w.r.t. true north, based on IMU yaw angle);
	-) set `qVM_x`, `qVM_y`, `qVM_z`, `qVM_w` based on the orientation of the external antenna with respect to your IMU;
	-) set `MrMV_x`, `MrMV_y`, `MrMV_z` based on the orientation of the external antenna with respect to your IMU;
-) Compile Rovio with NPOSE = 1 to co-estimat the transformation between the Rovio inertial frame (W) and the reference inertial frame (I) (see (Rovio wiki)[https://github.com/ethz-asl/rovio/wiki/Configuration] );
-) Start Rovio and redirect local ENU coordinates topic into rovio/pose topic. 

I hope this can help you.
Regards,
Marco.

bloesch, 2017-02-13:
For low-cost IMUs you should ideally increase the noise covariance, not reduce it. We recently did experiments with a BMI055 and increased the noise parameters of the accelerometer ("vel") and the gyroscope ("att") by a factor of 100. Obtaining higher drift for such a setup is just normal.

There are a couple of ideas around to extend rovio with a mapping back-end. One of the simplest would be to do bundle adjustment based mapping in a separate thread and to re-observe the corresponding landmarks in rovio.

greymfm, 2017-02-13:
Thanks. Actually I did increase covariance (sorry). What I do not understand is the unstable behavior (point 1. in my first post) after some time. Could this be related to covariance again, or are there other things I could try?

bloesch, 2017-03-12:
diificult to say what the issue is in the described case. the point is that rovio is not a 100% fail safe system. the easiest thing to do, but not supported yet, would be to detect such cases and reset/relocalize the rovio. this would not be a difficult thing to do but requires some work in terms of implementation.

otherwise tuning the parameter to your specific setting is always a good thing. unfortunately this requires quite some knowledge of the system. some hints are available under https://github.com/ethz-asl/rovio/wiki/Configuration#rovio-parameters

xiaopenghuang, 2017-03-25:
@bloesch 
> One of the simplest would be to do bundle adjustment based mapping in a separate thread and to re-observe the corresponding landmarks in rovio.

 how can we get the feature point 3d position (which i think the depth is not true), if 3d position is not right, How to transform a point to another image to minimize the reprojection error.

bloesch, 2017-03-30:
not sure I fully understand what you mean. the state of the filter stores the inverse distance (not depth). there should also be a pointcloud which is output to ROS.

The TransformFeatureOutputCT class takes care of projecting the landmarks into the different cameras. There is no explicit functionality to transform a 2d point from one image to the other. But given the distance this could be done manually.

closed 2018-01-11 by bloesch