ethz-asl/rovio #110: Image noise variance Parameters

isamabdullah88, 2017-02-13:
Hi, could anyone tell me what is the image noise variance parameter of the camera in the MH_01_easy dataset (ASL Dataset, Machine Hall 1)?
http://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets
Thanks

bloesch, 2017-02-13:
The parameters in the master branch are chosen in order to fit the euroc dataset - so you do not have to change anything. The corresponding ImgUpdate.UpdateNoise.int parameter is often difficult to derive from the camera specifications and often requires some extra tuning.

isamabdullah88, 2017-02-13:
Thanks for your answer. So I am actually trying to solve the problem with MSCKF and there I needed the camera/image noise variance parameter. Can you tell me what the exact value is/or where it is mentioned?
thanks a lot.

bloesch, 2017-03-12:
MSCKF uses a fundamentally different measurement model. it is not that easy to translate rovio's image intensity error model to MSCKF's reprojection error based metrics. i guess assuming a couple of pixel reprojection error is reasonable.

isamabdullah88, 2017-03-29:
Thanks alot.

isamabdullah88, 2017-03-29:
Hi, thanks for your answer. Can you comment on which models have obtained most accurate results? Are there similar works which do not involve kalman filters? In general how much accuracy has been achieved up till?

bloesch, 2017-03-30:
1. It is not that easy to rank the different models. Rovio uses intensity measurements directly and thereby avoid some issues related to standard feature tracking and matching. However, this induces some extra complexity to the system and reduces modularity. If there is no special motivation to got for direct approaches I would therefore recommend to stick with the classical reprojection error.

2. Most approaches use some information fusion techniques. This is typically either some form of Kalman filter or batch optimization (maximum likelihood estimation).

3. I do not feel that accuracy should be the primary focus of a SLAM front-end. But Rovio is able to compete with other state-of-the-art odometry approaches (the actual results are often very dataset and tuning dependent). If you want higher long-term accuracy you will not be able to avoid integrating some back-end mapping framework.

isamabdullah88, 2017-04-07:
Hi bloesch, I have successfully implemented MSCKF, but I have not got the accuracy they report. I think there is some fundamental theoretical inconsistencies in their measurement model. But before digging deep into them, I need to confirm them. So if you could provide an email, Can I  send some details of the inconsistencies/flaws to confirm?

bloesch, 2017-04-07:
I doubt that there are fundamental inconsistencies in their model and I know several people having applied MSCKF successfully.

isamabdullah88, 2017-04-07:
Well not fundamental, but there are some inconsistencies. I have myself implemented MSCKF successfully in my Final Year Project. But I think, removing those inconsistencies, would further increase the accuracy. I just need to confirm if they are really inconsistencies. If you give  me your email, I would send some of the details of that.

bloesch, 2017-04-12:
Feel free to write your thoughts here. Otherwise you can use bloeschm@ethz.ch.

isamabdullah88, 2017-04-13:
Hi, thanks for that.

closed 2017-09-19 by bloesch