ethz-asl/rovio #111: High altitude MAV state estimation

rikba, 2017-02-13:
Hi @bloesch ,

we use a VI-sensor and ROVIO for outdoor state estimation on our hexacopter. We fuse ROVIO's pose with the MAV's IMU into [msf](https://github.com/ethz-asl/ethzasl_msf) for position control.

When we were testing it outdoors, we observed some heavy shaking of our drone, especially when flying on higher altitudes (~10-15m). 

From the recorded dataset (with manual control) we could not really infer what was going wrong. The position and angle measurements seem to be smooth and there were many tracked features.

Our first guesses were:
- ROVIO puts too much trust into features which are too far away,
- the VI-IMU was too noisy,
- distant features require more computational power leading to a delay in ROVIO.

If you want to reproduce our estimates, I sent you the VI-Sensor (cam0, imu0) and onboard imu (imu) bag of a manual flight.

Maybe you have some idea what we can do to start debugging. We will point the sensor more downwards, now, and dampen the VI-IMU more.

Best,
@rikba

bloesch, 2017-02-13:
thanks guys, life would get boring here without you =) I will have a look at it until tomorrow.

bloesch, 2017-02-13:
I had a quick look. Seems to work fine. However scenes like this where you are further away from the scene are always quite tricky and I would recommend using GPS. Do you have GPS data for that bag?

rikba, 2017-02-14:
Thanks a lot. We will fuse GPS next time. And we will have the camera more downward facing. There certainly is a demand for @bloesch here.

fmina, 2017-02-15:
#68 also report the same issue. @rikba the discussion is interesting 🚁

rikba, 2017-02-16:
Trusting ROVIO less and onboard IMU more already helps quite a bit up to certain height. Everything else is adressed in #68 .

Thanks again!

closed 2017-02-16 by rikba