ethz-asl/rovio #112: Jumps between world and map when NPOSE=1

marco-tranzatto, 2017-02-13:
@bloesch : we can't survive one week without you!
@michaelpantic and I have still some very strange "jumps" between world and map frames when using [MalfunctioningPoseUpdate](https://github.com/ethz-asl/rovio/tree/fix/MalfunctioningPoseUpdate) brench with external GPS data.

I sent you our bag using CIFEX .... if you have some time could you help us please?
Main issues:

- Rovio odometry jumps two times, around 27.2 sec and 29.5 sec. This happens even tough the GPS position neither jumps nor drops out.
- We spent the whole day debugging it and we found out something strange about the H matrix. We believe there's something fishy [here](https://github.com/ethz-asl/rovio/blob/fix/MalfunctioningPoseUpdate/include/rovio/PoseUpdate.hpp#L258). Incredibily the filter does not jump if we hard code `F.template block<3,3>(mtInnovation::template getId<mtInnovation::_pos>(),mtState::template getId<mtState::_poa>(inertialPoseIndex_)) = M3D::Identity();`

Please add a 'Donate Now' button with paypal, so we can finance your coffees while you debug Rovio :D

michaelpantic, 2017-02-14:
Just some further info:

- It seems that shortly before the jumps, the update vector norm (in lighweight

michaelpantic, 2017-02-14:
Ups, accidentaly closed, reopened...

bloesch, 2017-02-14:
The position update uses the following scheme: 
    // IrIV = IrIW + qWI^T*(WrWM + qWM*MrMV)
I believe that the issue could be related to the growing leverarm given by WrWM, which increases the values in the Jacobian w.r.t. qWI and consequently the sensitivty of qWI as well. Could briefly doublecheck if this is correct, i.e., you get shaky estimates when WrWM is large and not otherwise? Also is this jitter only on qWI, or can you observe it on other IrIW as well?

I knew that this could happen but was expecting that WrWM needs to be really large (it will be covariance parameter dependent as well). If this is indeed the origin of the problems I see three approaches:
- Tune the covariances in order to decrease the jitter (decrease pre_cov_qWI).
- Change the parametrization to use WrIW instead of IrIW. This would allow to use a moving I frame which follows the sensor in order to reduce the leverarm. This would require some coding and new testing.
- Do the control based on WrWM instead of IrIM and transform your desired location from I to W with a lower frequency.

It would be nice if you could quickly doublecheck the above hypothesis.

bloesch, 2017-03-12:
any updates on the issues?

marco-tranzatto, 2017-04-11:
Hi @bloesch ! 

Sorry for the incredible delay, but we couldn't investigate this issue anymore in February and March. Nevertheless we had great results at MBZIRC with NPOSE=0 :)

Now I have finally some time to spend on it and I'd really like to figure out what the problem is.

I sent you another data-set using CIFEX, where the "shaky" behavior happens again, this time a few seconds after takeoff.
Data-set setup:
- map frame, inertial (I): placed in the center of an arena, where the MAV was flying
- world frame, odometry (W): initialized by rovio based on IMU orientation and GPS position: IrIW = [97.907, 34.294, -0.271] at the beginning of the bag (obtained with command `rosrun tf tf_echo map world`)

Requirements:
- use "wolrd" as fixed frame in Rviz, otherwise shakes will not be visible;
- use branch [MalfunctioningPoseUpdate](https://github.com/ethz-asl/rovio/tree/fix/MalfunctioningPoseUpdate), compiled with `-DROVIO_NPOSE=1`
- play bag with `rosbag play only-rovio-input-2017-03-14-08-35-31.bag --clock`

I investigate a little bit the points you mentioned.

> Could briefly double check if this is correct, i.e., you get shaky estimates when WrWM is large and not otherwise? Also is this jitter only on qWI, or can you observe it on other IrIW as well?

- I get shaky estimates when WrWM is "small", i.e. when the MAV is close to the origin of world, around 34 seconds after the bag has started.
Before the shake happens, I have the following position vector: WrWM = [0.002, -7.404, 5.191] (obtained with command `rosrun tf tf_echo world imu`) ... the euclidean distance is around 9 meters, this is definitely not "big" enough;
- It does not seem like IrIW moves too much: by using again the command `rosrun tf tf_echo map world 10` (this time it updates at 10 Hz), IrIW stays more or less where it was before the jumps. 

> Tune the covariances in order to decrease the jitter (decrease pre_cov_qWI).
- I tried to decrease `pre_cov_qWI` from 1e-4 to 1e-6 (odometry still jumps around 35 seconds), to 1e-8 (odometry still jumps around 35 seconds), to 1e-10 (odometry still jumps around 35 seconds);

> Change the parametrization to use WrIW instead of IrIW. This would allow to use a moving I frame which follows the sensor in order to reduce the leverarm. This would require some coding and new testing.
- I'm not really sure how to do it and if I should try to do it :scream: 

> Do the control based on WrWM instead of IrIM and transform your desired location from I to W with a lower frequency.
- I believe the control already uses WrWM, in fact the odometry message output by rovio (and subscribed by the controller) is obtained from WrWB: [odometry publisher](https://github.com/ethz-asl/rovio/blob/fix/MalfunctioningPoseUpdate/include/rovio/RovioNode.hpp#L693)

Thanks again for your help!

Marco.

closed 2018-01-11 by bloesch